adafruit/circuitpython #3063: Eject USB MSC from device side?

xiongyihui, 2020-06-23:
Right now, CircuitPython can't remount the MSC storage when it has been mounted by the computer. We can add a feature to  eject the storage from the device side. The feature is already supported by [DAPLink](https://github.com/armmbed/daplink/blob/master/source/usb/msc/usbd_msc.c#L520).

tannewt, 2020-06-23:
It isn't possible to eject from the device side as far as I know. We could simply make the drive disappear but that is very risky since it is common for the host OS to cache filesystem changes.

What are you trying to achieve?

xiongyihui, 2020-06-24:
To add a new file to the storage without user .

It is possible. DAPLink supports drog-n-drop programming which is similar to uf2 programming. When something gets wrong, DAPLink will create a text file with a error message.

xiongyihui, 2020-06-24:
The host OS will drop any cache when a device is ejected.

tannewt, 2020-06-24:
> To add a new file to the storage without user .

I'm not sure what you mean by this. If you want to write to the filesystem, you can do it before USB is initialized when CircuitPython starts up.

> It is possible. DAPLink supports drog-n-drop programming which is similar to uf2 programming. When something gets wrong, DAPLink will create a text file with a error message.

I don't think DAPLink actually has a FAT filesystem. (UF2's filesystem is fake.)

With CircuitPython, the FAT filesystem is maintained by the host because mass storage is a block level protocol. The writes to the blocks from the host don't ensure a valid filesystem structure unless the host cache has been flushed. The device shouldn't disappear too soon.

We could consider swapping filesystem writability if the host ejects the device. However, "eject" means different things on different OSes so it may not work across all of them.

xiongyihui, 2020-06-24:
Following the guide - https://learn.adafruit.com/circuitpython-essentials/circuitpython-storage,  the device is able to write a new file to storage, but we need to reboot the device to see the file. It's a solution anyway.
I agree the cache issue that two writers may conflict. When the storage is read-only to the host, ejecting from device side cloud work.

closed 2020-06-24 by xiongyihui