adafruit/circuitpython #3066: USB: Check number of descriptors at build time

jepler, 2020-06-24:
In #3038 @dhalbert noted it would be handy to check that the enabled USB devices fit within the MCU-specific number of USB descriptors:
> Another check we could add (this is getting complicated) is counting endpoints for boards with limited endpoints, and warning you there are too many. It has to know CDC takes 2, and others take only one. It could count words/commas in USB_DEVICES.

As it stands, I think such builds succeed but crash when loaded on a board.

jepler, 2021-03-15:
@dhalbert didn't you add this checking to all platforms when you added the 2nd CDC?  If so, please close.

dhalbert, 2021-03-15:
I didn't even know there was an issue :). FIxed by #4215.

closed 2021-03-15 by dhalbert