adafruit/circuitpython #3075: ESP32-S2 not recognized on Windows

PTS93, 2020-06-26:
Using the Saola-1 Wroom devkit and running CircuitPython 6.0 alpha or 5.4.0 beta or really any CPY version that I tried I can't get the device to enumerate on Windows. On a Ubuntu 18.04 machine it shows up as a MSC drive and I can connect to REPL.
On Windows I just get an error that the USB device malfunctioned.

I would assume that this is an issue with TinyUSB and not CPY but I wanted to open it her in case it is not. I have not verified if TinyUSB standalone would work.
@hathach

hathach, 2020-06-26:
@PTS93  can you verify that it work with bare metal example from tinyusb here 
https://github.com/hathach/tinyusb/tree/master/examples/device/cdc_msc_freertos

just run `idf.py build flash`

PTS93, 2020-06-26:
Ok this is odd. Now that I formatted the CPY drive in Linux through the REPL it now shows up on Windows too. Maybe that was the issue that the FAT device was not properly formatted and windows is picky there? Or I'm having an issue with the connector here. It was picky with Linux too when I used a breadboard for the native USB.

PTS93, 2020-06-27:
I took a closer look at my bread board. Turns out I chose the exact point where the breadboard had a short which caused D- to short to VBUS. It's an intermittent short and I guess Linux was more tolerant to that.
[image]

closed 2020-06-27 by PTS93