adafruit/circuitpython #3076: circuitpython-ulab update

khashdroid, 2020-06-26:
Hi there!
I have a short question. I am using circuitpython & ulab very often. I noticed that the ulab used in circuitpython is 4 months old. I was wondering if the circuit-python ulab will be updated anytime soon to match the micropython-ulab, since I need to use some of the recent developments from ulab (specifically the newly added sosfilt function, which is much needed in DSP applications)
 
Thanks in advance.

jepler, 2020-06-26:
Hi!  We would love community help staying up to date with ulab.

Right now, there are a couple of steps to doing this:
 * Actually update the ulab submodule in `extmod/ulab`
 * Update our ulab documentation in `shared-bindings/ulab`
 * Build and test, especially that new and changed functionality is working
 * Create a PR (pull request)
 * Work through any problems identified during the PR process

If you have the technical expertise to do this, we would love to see a pull request from you!

jepler, 2020-06-26:
If you don't have the technical expertise to do this _yet_ but would like to learn, we would love to help you grow, so feel free to ask us about it here or on discord in #circuitpython.

tannewt, 2020-07-13:
Updated 12 days ago to 0.51.1

closed 2020-07-13 by tannewt