adafruit/circuitpython #3077: autodocs switch broke links from libraries and "latest" link from other docs versions

siddacious, 2020-06-26:
Commit 4e8de3c554024d3ac89e9ee77b958a6932a64bec to switch to autoapi changed the path to as far as I can tell all of the docs from

https://circuitpython.readthedocs.io/en/latest/shared-bindings/busio/I2C.html#busio.I2C to
https://circuitpython.readthedocs.io/en/latest/autoapi/busio/index.html#busio.I2C

This broke all of the links to core classes from libs, as well as the 'latest' link from the non-latest versions, as well as the links to the non-latest versions from latest

jepler, 2020-06-26:
this was supposed to be improved by #3031 , did I miss the mark?

jepler, 2020-06-26:
Is it possible that the transition to the "main" branch means that "latest" is not reflecting the newest changes to "main"?

tannewt, 2020-06-26:
Yup, let me check.

tannewt, 2020-06-26:
Ya, it wasn't updated. RTD has a default branch setting too. I've triggered a new latest build.

jepler, 2020-06-26:
The redirects aren't working on readthedocs:
> WARNING:root:The 'sphinxcontib-redirects' plugin is only supported by the 'html' builder. Skipping...

https://readthedocs.org/api/v2/build/11338560.txt

The build is invoked with the argument `-b readthedocs` and not `-b html`.  The redirects-builder checks for exactly the html builder.  https://github.com/readthedocs/readthedocs-sphinx-ext/blob/master/readthedocs_ext/readthedocs.py#L316  https://github.com/adafruit/circuitpython/blob/main/conf.py#L397

We (I) should change the check to not be for an exact type, but for a sub-type too.

closed 2020-06-27 by tannewt

jepler, 2020-06-27:
I checked that for all the following URLs, going from "5.0.x" to "latest" gives a working link:
https://circuitpython.readthedocs.io/en/5.0.x/shared-bindings/
https://circuitpython.readthedocs.io/en/5.0.x/shared-bindings/support_matrix.html
https://circuitpython.readthedocs.io/en/5.0.x/shared-bindings/_bleio/__init__.html
https://circuitpython.readthedocs.io/en/5.0.x/shared-bindings/_bleio/Address.html
https://circuitpython.readthedocs.io/en/5.0.x/shared-bindings/audiomp3/__init__.html

Going in the opposite direction does not work, e.g., from
https://circuitpython.readthedocs.io/en/latest/shared-bindings/audiomp3/
going to 5.0.x gives a 404 page.  We _could_ potentially add redirects in 5.0.x that go in the reverse direction.  I can work on this if it is desired, but it should be considered a separate issue.

PR #3078 (2020-06-27): Make redirects actually work on RTD, redirect __init__.html pages
Closes #3077