adafruit/circuitpython #3084: CMake version requirement will not build esp32s2 on Raspberry pi

jerryneedell, 2020-06-27:
The version of CMake required for the esp32s2 build was set to >=3.16 but Raspberry Pi's running Raspbian Buster are currently using CMake v3.13 so the esp32s2 build fails.

I tried setting the required version to >=3.13 and it build OK.

Is there any objection to reverting this to 3.13 

It was set to 3.16 by PR #3072

jepler, 2020-06-27:
The version of cmake _SHOULD_ be the one installed when you use the esp-idf setup process, and it should always be 3.16.
```
$IDF_PATH/tools/idf_tools.py --non-interactive install cmake
source $IDF_PATH/export.sh
```
After these commands, which are used during the github actions build process, the required version of cmake should be available AND placed first on the path.  If this is not the case, it should probably be pursued as a bug upstream at esp-idf.

```
$IDF_PATH/tools/idf_tools.py --non-interactive install cmake
ERROR: tool cmake does not have versions compatible with platform linux-armel
```

jepler, 2020-06-27:
Or is it the case that this is literally the only thing preventing the build from working?  If so, maybe a workaround should be used.  But it might break at any moment if they add features provided by the version of cmake that they provide for normal desktop systems.

jerryneedell, 2020-06-27:
sorry -- I added the label by accident -- how do I get rid of it?

jerryneedell, 2020-06-27:
> The version of cmake _SHOULD_ be the one installed when you use the esp-idf setup process, and it should always be 3.16.
>
> ```
> $IDF_PATH/tools/idf_tools.py --non-interactive install cmake
> source [...]

i'm sorry -- I cant't figure out where this is coming from.

jepler, 2020-06-27:
These commands (among others) are run during the normal github build process for esp32s2, they are in .github/workflows/build.yml:
```
$IDF_PATH/tools/idf_tools.py --non-interactive install cmake
source $IDF_PATH/export.sh
```
IDF_PATH has been set to the full path of the esp-idf directory inside ports/esp32s2.

However, running that on a pi (I tried on a pi 4) gives the error message I showed:
```
ERROR: tool cmake does not have versions compatible with platform linux-armel
```

I don't exactly know why our build script does one thing and the online docs show another thing.

jerryneedell, 2020-06-27:
> These commands (among others) are run during the normal github build process for esp32s2, they are in .github/workflows/build.yml:
>
> ```
> $IDF_PATH/tools/idf_tools.py --non-interactive install cmake [...]

OK -- now I see it in the GitHub workflow  -- I have just ben doing 
```./install.sh
. ./export.sh
``` in the esp-idf folder 
I guess I'm not sure where or why the make 3.16 requirement came from

jerryneedell, 2020-06-27:
This not a big deal -- I have a workaround I can use and I guess no one else uses a Pi -- feel free to close until it becomes a problem for someone.  I only use the Pi as a "backup"

jerryneedell, 2020-06-27:
In the EDP-IDF instructions cmake is installed as a prerequisite using "apt-get make"  so on the Pi it get version 3.13.4. I "could" download and build 3.16, but since 3.1.3 works OK it seemed simple to just use it.

the notes in #3072 only indicated that it had to be above 3.10 for some reason -- it looked like 3.16 was an arbitrary choice.

jepler, 2020-06-27:
Every CMakeLists.txt in esp-idf that has a version line says `CMakeLists.txt:cmake_minimum_required(VERSION 3.5)` so maybe we should back ours down.  I don't know where a requirement for 3.10 or 3.16 comes from.

jepler, 2020-06-27:
```$ grep minimum docs/en/api-guides/build-system.rst  
      cmake_minimum_required(VERSION 3.5)
- ``cmake_minimum_required(VERSION 3.5)`` tells CMake the minimum version that is required to build the project. ESP-IDF is designed to work with CMake 3.5 or newer. This line must be the first line in the CMakeLists.txt file.
```
especially since they document this as being the case.

jepler, 2020-06-27:
Some things about how _WE_ invoke cmake do require something newer than 3.5  There is some confusion about -S and -B flags specified in the order we use, but more importantly 3.5 does not know about Ninja files.

However, since esp-idf says 3.5, we can set it to the minimum of what we require and what they require.  So why not go ahead and pull request to change it to 3.10, or 3.12, whatever makes more sense with you.  If we won't _continually_ verify 3.10 compatibility, specifying something you think you need is probably better.

jerryneedell, 2020-06-27:
Ok, I’ll put one in for 3.13  -won’t be until tomorrow though.

PR #3086 (2020-06-28): change esp32s2 CMake verision requirement to >= 3.13
fixes #3084 
Raspberry Pi Raspbian Buster uses CMake v 3.13.4  - this allows building esp32s2 on Raspberry Pi.

Tested on Raspberry Pi 4 B +
closed 2020-06-28 by jepler