adafruit/circuitpython #3091: Broken links from Autodocs switch still remain

siddacious, 2020-06-29:
Same core issue as #3077

While on this page:
https://circuitpython.readthedocs.io/en/latest/shared-bindings/board/
The "5.0" version link points to 

https://circuitpython.readthedocs.io/en/5.0.x/shared-bindings/board/index.html  
which does not exist. The actual URL is 
https://circuitpython.readthedocs.io/en/5.0.x/shared-bindings/board/__init__.html

To be explicit about the scope of this issue, I won't consider it fixed until all known broken links that resulted from the autodocs switch have been fixed.

jepler, 2020-06-29:
Thanks for filing a fresh github issue, even though it's the same underlying problem.  The way github treats linked pull requests & issues, combined with the way that I don't know how to "do a test" to make sure everthing is fixed, means we may end up spending a lot of small integers while working on this problem.

tannewt, 2020-07-13:
@jepler this fixed?

jepler, 2020-07-14:
No, there's been no action taken on this issue.  This problem relates to following "latest" links back to old versions, and requires adding redirects to 5.x and older versions so that the "new" URLs work.  The general mold of what is on main can be followed, but the "from" and "to" need to be swapped.

dhalbert, 2023-10-28:
Fixed at some point.

closed 2023-10-28 by dhalbert