adafruit/circuitpython #3098: Binding littleVGL

CaptainJackey, 2020-07-01:
Will CircuitPython has plan to binding littleVGL ?

tannewt, 2020-07-01:
We don't have plans to add it. If someone made a PR I'd be happy to merge but would want to make sure the API is pythonic (attributes instead of getters/setters for example) instead of a straight-C wrapper.

closed 2020-07-01 by tannewt

CaptainJackey, 2020-07-02:
> We don't have plans to add it. If someone made a PR I'd be happy to merge but would want to make sure the API is pythonic (attributes instead of getters/setters for example) instead of a straight-C wrapper.

LVGL‘s team grow quickly , and I think it world be the best fit GUI for micropython ,  actually they had finish the python objects and has full samples of all widgets. but still less data about micropython.

https://github.com/lvgl/lv_examples/tree/master/src/lv_ex_widgets