adafruit/circuitpython #3110: WeChat group?

tannewt, 2020-07-02:
Hi @dahanzimin, @ryevertigo, @namaRyo, and any other Chinese devs, would you be interested in starting a WeChat group for CircuitPython? I can sort out how to get an account. Thanks!

notelainemade, 2020-07-03:
Hello Scott @tannewt, this is Elaine from Seeed! We just finished the LCD driver added into Wio Termina today :)  I will be glad to join the WeChat group to share more CircuitPython dev process for the community in China. May I add your WeChat if you already get the account?

tannewt, 2020-07-03:
Hi Elaine! I don't have an account yet but you can always email me scott@adafruit.com too. I'll let you know when I have a WeChat account.

tannewt, 2020-07-03:
Ok, I'm `tannewt` on WeChat now! 🎉

namaRyo, 2020-07-03:
Hi, my name is Ryo Ikuyama. my weChat id is "ikuyama"
please invite me if I can join.

notelainemade, 2020-07-03:
My wechat id is "iamElaineee"!

notelainemade, 2020-07-03:
@tannewt  Sent request

ptorrone, 2020-07-03:
@tannewt please add `adafruit` when ya can too! (that's our wechat id)

hierophect, 2020-07-03:
Hey all. Would anyone be willing to help verify my account? Would love to help with any stm32 discussions.

tannewt, 2020-07-03:
@namaRyo I invited you as a friend and can add you to the group chat after that.

dahanzimin, 2020-07-13:
> 大家好。有人愿意帮我核实一下我的账户吗？愿意帮助任何stm32讨论。

@hierophect What is your wechat? May I add you to discuss the STM32 series

hierophect, 2020-07-13:
Hi @dahanzimin, I have not been able to get a wechat because I do not know any mainland Chinese people who can help me with security verification. Would you be interested in helping me verify my account? I would love to discuss the STM32.

dahanzimin, 2020-07-13:
> Hi @dahanzimin, I have not been able to get a wechat because I do not know any mainland Chinese people who can help me with security verification. Would you be interested in helping me verify my account? I would love to discuss the STM32.

I'm very happy. I understand that you need to download the wechat app and register with your mobile phone number. Of course, I haven't registered wechat overseas, or how can I help you complete the security certification

dahanzimin, 2020-07-13:
> Hi @dahanzimin, I have not been able to get a wechat because I do not know any mainland Chinese people who can help me with security verification. Would you be interested in helping me verify my account? I would love to discuss the STM32.

@hierophect I Baidu, need security authentication QR code, you send me email, I help you authentication, 353767514@qq.com

hierophect, 2020-07-13:
[image]
@dahanzimin

tannewt, 2020-07-13:
@dahanzimin I'm on WeChat as `tannewt` and we have a CircuitPython group setup I'll invite you to.

tannewt, 2020-07-14:
Closing. Folks can add me on WeChat and I'll add you to the group.

closed 2020-07-14 by tannewt