adafruit/circuitpython #3112: Object init error string shows wrong number of arguments

dunkmann00, 2020-07-03:
When initializing an object, the error that is produced when the wrong number of positional arguments is given shows the wrong number of arguments given and required. It is off by +1. From my testing, this only happens when the object being initialized is a subclass.

Example:
```python
import board
import neopixel # subclass of either _pixelbuf or pypixelbuf, depending on CP version and/or board

extra_arg = None

pixels = neopixel.NeoPixel(board.D5, 24, extra_arg) # This is supposed to have only 2 positional arguments here
```

Output:
```python
TypeError: function takes 4 positional arguments but 5 were given
```

Expected Output:
```python
TypeError: function takes 3 positional arguments but 4 were given
```

I tested this on an itsybitsy_m4_express running CP 5.3.0 and 5.4.0-beta.1

dhalbert, 2020-07-03:
This is not a bug. CPython works the same way. You count the number of arguments in the function definition, not in the parentheses at the point of call. So the argument count includes `self`. E.g. `self.func(2)` is two arguments, `self` and `2`.

```python
class C:
    def __init__(self, x):    # 2 arguments
        self.x = x
```
CPython:
```python
Python 3.8.2 (default, Apr 27 2020, 15:53:34) 
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from args import C
>>> c = C(3, 4)    # 3 arguments
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: __init__() takes 2 positional arguments but 3 were given
>>> 
```

CircuitPython:
```python
Adafruit CircuitPython 6.0.0-alpha.0-6-g11cb3e3b4-dirty on 2020-07-01; Adafruit Metro M4 Airlift Lite with samd51j19
>>> from args import C
>>> c = C(2, 3)    # 3 arguments
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: function takes 2 positional arguments but 3 were given
>>> 
```

dhalbert, 2020-07-03:
Sorry, i misunderstood and posted an incorrect response, which I've deleted.

dhalbert, 2020-07-03:
Reproduced. I also tried a third layer of class nesting (not shown), and got the same off-by-1 error (as opposed to, say, off by 2).

This bug is also present in MicroPython, so I'll file an issue there.

```python
class A:
    def __init__(self):
        pass

class B(A):
    def __init__(self):
        super().__init__()

try:
    print("A(1)")
    A(1)
except Exception as e:
    print(e)

try:
    print("B(1)")
    B(1)
except Exception as e:
    print(e)
```

```python
Adafruit CircuitPython 6.0.0-alpha.0-6-g11cb3e3b4-dirty on 2020-07-01; [...]
>>> import args
A(1)
function takes 1 positional arguments but 3 were given
B(1)
function takes 2 positional arguments but 3 were given
>>> 
```

dunkmann00, 2020-07-04:
Sounds good!