adafruit/circuitpython #3125: `time.time()` returns int rather than float

Flameeyes, 2020-07-06:
In CPython 3.8 `time.time()` returns a floating point value. In CircuitPython 5.3 it returns `int` (or rather a long integer).

`time.monotonic()` returns `float` though.

Flameeyes, 2020-07-06:
It appears this is behind a `#define`:

```
STATIC mp_obj_t mod_time_time(void) {
#if MICROPY_PY_BUILTINS_FLOAT
    struct timeval tv;
    gettimeofday(&tv, NULL);
    mp_float_t val = tv.tv_sec + (mp_float_t)tv.tv_usec / 1000000;
    return mp_obj_new_float(val);
#else
    return mp_obj_new_int((mp_int_t)time(NULL));
#endif
}
```

dhalbert, 2020-07-06:
```c
STATIC mp_obj_t mod_time_time(void) {
```
This is only in ports/unix/modtime.c. The erroneous code returning an `int` is in
`shared-bindings/time/__init__.c`.

We do assume all CircuitPython impls have floats (but not longints).

dhalbert, 2025-01-31:
A similar issue recently came up in https://github.com/adafruit/Adafruit_CircuitPython_datetime/issues/29 and https://github.com/adafruit/Adafruit_CircuitPython_datetime/pull/30. It is better to use an int because the precision of the 30-bit floats is so limited. I think we will not fix this for the reasons given in https://github.com/adafruit/Adafruit_CircuitPython_datetime/pull/30.

closed 2025-01-31 by dhalbert