adafruit/circuitpython #3126: "ValueError: All timers for this pin are in use" for pin that supports PWM and is not being used for anything else.

AdinAck, 2020-07-07:
I have an ItsyBitsy M4 and am trying to use pin A5 as a pwm output however I am getting this error.
Here is the line that errors:
`fan = pulseio.PWMOut(board.A5, frequency=5000, duty_cycle=0)`
A5 is not used for anything else so that shouldn't be an issue.

Thanks!

deshipu, 2020-07-07:
Timers are shared between several pins. It possible all the timers that pin A5 can use are already used by other pins.

AdinAck, 2020-07-07:
I have no other PWM pins in use, I have an SPI and I2C bus set up, I don't know If those count as PWM or require timers but they are the only other pins that are used I can think may interfere.

deshipu, 2020-07-07:
I would try using PWM on another pin.

AdinAck, 2020-07-07:
Unfortunately, I cannot use another pin because I am using the ItsyBitsy as an embedded microcontroller for a PCB I made. 

On a side note, I tried initializing the PWM pin before anything else (any digitalio/analogio/busio) and it still errored.
It's not like it errored later, it errored on the line of code initializing the PWMOut.

deshipu, 2020-07-07:
On ItsyBitsy M4, pin A5 is PA06. Looking at the datasheet for SAMD51, only ``TC1/WO[0]`` is available on that pin. I wonder if that timer is used internally for something.

wallarug, 2020-07-07:
The TCx/WO[0] timers are used for some special function (floating point maths or something) and are not usable as PWM or other functions.

dhalbert, 2020-07-07:
Not floating point, but #1766 I believe is talking about the fact that we can only use one capture channel per TC if we want 16-bit frequency accuracy, because of the way the TC registers are used for different functions. So we cannot use some TC channels despite them appearing to be available. @tannewt may remember the details.

tannewt, 2020-07-07:
WO[0] on the TCs is used for the top value in 16-bit mode. That gives us much finer control over the PWM frequency.

If you want to make it more advanced, then you could have it work but with coarse frequency. What are you using it for?

AdinAck, 2020-07-07:
I am just using it to control a small fan, nothing fancy.
I don't understand anything you guys are saying but it sounds like you are saying that pin A5 cannot do PWM, if that is the case you probably should update the pinouts section for the ItsyBitsy M4 for it says, "A4 and A5 have PWM output.".

AdinAck, 2020-07-08:
So...is there anything I can do to use at least very rudimentary PWM on this pin?

wallarug, 2020-07-08:
> So...is there anything I can do to use at least very rudimentary PWM on this pin?

I haven't tried this but I had an idea (probably a bad one).   

What if you DigitalWrite() (change pin.value) many many times in a while loop?

Don't know how fast you can switch it that way.  There are probably some extra hardware / circuit requirements as well.

AdinAck, 2020-07-08:
Yeah, that's a good idea but I can't do that because my loop has many things in it that take a lot of time (multiple seconds) :(

wallarug, 2020-07-08:
I guess you are out of options for using just one board 😞 

You might just have to get another expansion chip that can do the PWM function for you or a small Arduino and send it the commands over serial or something.  In the short term, WO[0] is not going to be freed up.

tannewt, 2020-07-08:
> I am just using it to control a small fan, nothing fancy.
> I don't understand anything you guys are saying but it sounds like you are saying that pin A5 cannot do PWM, if that is the case you probably should update the pinouts section for the ItsyBitsy M4 for it says, "A4 and A5 have PWM output.".

The pinout is for both Arduino and CircuitPython and with Arduino you can do PWM on that pin. With CircuitPython you cannot because our API allows you to choose PWM frequency.

There are two ways to fix this 1) change the software to work (switch to Arduino or modify CircuitPython) or 2) change the hardware. Both have trade-offs. I'm happy to guide you modifying CircuitPython but it isn't work I'll personally do soon.

dhalbert, 2020-07-08:
I have edited the Learn Guide to clarify which pins are not available in CircuitPython for PWM. (I just tried a bunch.)

AdinAck, 2020-07-09:
I'll just use it as a standard GPIO then, thanks for updating the learn guide.
Should I leave this issue open for (possibly very far) future implementation?

wallarug, 2020-07-09:
Nope.  You should close this one.  Issue #1766 has you covered!  Just follow that one 😉

tannewt, 2020-07-09:
Closing in favor of #1766. Thanks @AdinAck and @wallarug

closed 2020-07-09 by tannewt