adafruit/circuitpython #3128: Requesting USB VID/PID for the Raytac MDBT50Q-DB-40

Arudinne, 2020-07-07:
I was told that I cannot reuse the VID/PID in this pull request: https://github.com/adafruit/circuitpython/pull/3127

I do not have my own pair and I do not see anything about one in Raytac's documentation for this board: https://www.raytac.com/product/ins.php?index_id=81

Thanks!

dhalbert, 2020-07-07:
Here you go!
```
RayTac MDBT50Q-DB-40  VID 0x239A    PID 0x00BB  # bootloader
                                    PID 0x80BB  # arduino
                                    PID 0x80BC  # circuitpython
```

For CircuitPython, use these values.
```
USB_VID = 0x239A
USB_PID = 0x80BC
```

closed 2020-07-07 by dhalbert

markbirss, 2021-11-26:
@dhalbert 
Hi Dan

Does this mean I can use this for the Adafruit Bootloader for this same board
https://github.com/adafruit/Adafruit_nRF52_Bootloader/pull/233

```
USB_VID = 0x239A
USB_PID = 0x00BB
```