adafruit/circuitpython #3133: ESP32-S2 saola wrover Linux Cannot access drive.

palmtreefrb, 2020-07-09:
Ubuntu 18.04
Fresh flash using adafruit-circuitpython-espressif_saola_1_wrover-en_US-6.0.0-alpha.1.bin

From mu editor
Starting Mu 1.0.3
2020-07-08 18:50:34,317 - root:113(run) INFO: uname_result(system='Linux', node='fredrick-Studio-XPS-435MT', release='5.3.0-62-generic', version='#56~18.04.1-Ubuntu SMP Wed Jun 24 16:17:03 UTC 2020', machine='x86_64', processor='x86_64')
2020-07-08 18:50:34,317 - root:114(run) INFO: Python path: ['/home/fredrick/Development/circuitpython/ESP32-S2/venv/bin', '/usr/lib/python36.zip', '/usr/lib/python3.6', '/usr/lib/python3.6/lib-dynload', '/home/fredrick/Development/circuitpython/ESP32-S2/venv/lib/python3.6/site-packages', '/home/fredrick/Development/circuitpython/ESP32-S2/venv/lib/python3.6/site-packages/IPython/extensions']
2020-07-08 18:50:34,317 - root:115(run) INFO: Language code: en_US
2020-07-08 18:50:34,391 - mu.logic:543(__init__) INFO: Setting up editor.
2020-07-08 18:50:34,391 - mu.logic:562(__init__) INFO: Settings path: /home/fredrick/.local/share/mu/settings.json
2020-07-08 18:50:34,391 - mu.logic:563(__init__) INFO: Session path: /home/fredrick/.local/share/mu/session.json
2020-07-08 18:50:34,391 - mu.logic:564(__init__) INFO: Log directory: /home/fredrick/.cache/mu/log
2020-07-08 18:50:34,391 - mu.logic:565(__init__) INFO: Data directory: /home/fredrick/.local/share/mu
2020-07-08 18:50:34,395 - mu.logic:578(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero
2020-07-08 18:50:34,481 - mu.logic:1150(change_mode) INFO: Workspace directory: /home/fredrick/mu_code
2020-07-08 18:50:34,482 - mu.logic:627(restore_session) INFO: Restoring session from: /home/fredrick/.local/share/mu/session.json
2020-07-08 18:50:34,482 - mu.logic:628(restore_session) DEBUG: {'theme': 'day', 'mode': 'adafruit', 'paths': [], 'envars': [], 'minify': False, 'microbit_runtime': '', 'zoom_level': 2}
2020-07-08 18:50:34,482 - mu.logic:649(restore_session) INFO: Loaded files.
2020-07-08 18:50:34,482 - mu.logic:653(restore_session) INFO: User defined environment variables: []
2020-07-08 18:50:34,482 - mu.logic:657(restore_session) INFO: Minify scripts on micro:bit? False
2020-07-08 18:50:34,520 - mu.logic:680(restore_session) INFO: Starting with blank file.
2020-07-08 18:50:34,532 - mu.logic:1150(change_mode) INFO: Workspace directory: /media/fredrick/CIRCUITPY
2020-07-08 18:50:45,902 - mu.logic:1077(show_admin) INFO: Showing logs from /home/fredrick/.cache/mu/log/mu.log

[image]

palmtreefrb, 2020-07-09:
More info. Trying to access from terminal...

fredrick@fredrick-Studio-XPS-435MT:/media/fredrick/CIRCUITPY$ sudo su
[sudo] password for fredrick: 
root@fredrick-Studio-XPS-435MT:/media/fredrick/CIRCUITPY# ls
**ls: cannot access 'lib': Input/output error**
boot_out.txt  lib
root@fredrick-Studio-XPS-435MT:/media/fredrick/CIRCUITPY#

jerryneedell, 2020-07-09:
@palmtreefrb I found on mine, after a fresh build/install,  I had to reformat the internal dive.  If you open a terminal section to the board and at the REPL type
 ```
import storage
storage.erase_filesystem()
``` 
it will regenerate the file system.

palmtreefrb, 2020-07-09:
After running erase_filesystem() I can now mount and access the directories. 
dmesg still reports some errors though...

root@fredrick-Studio-XPS-435MT:/media/fredrick/CIRCUITPY# sudo dmesg
[38447.024174] usb 3-2.3: USB disconnect, device number 11
[38455.438884] usb 3-2.3: new full-speed USB device number 12 using xhci_hcd
[38455.561785] usb 3-2.3: New USB device found, idVendor=239a, idProduct=80a6, bcdDevice= 1.00
[38455.561787] usb 3-2.3: New USB device strings: Mfr=2, Product=3, SerialNumber=1
[38455.561788] usb 3-2.3: Product: Saola 1 w/WROVER
[38455.561789] usb 3-2.3: Manufacturer: Espressif
[38455.561789] usb 3-2.3: SerialNumber: 0000000000000000350000000CFCDF
[38455.666949] cdc_acm 3-2.3:1.0: ttyACM0: USB ACM device
[38455.675438] usb-storage 3-2.3:1.2: USB Mass Storage device detected
[38455.675542] scsi host12: usb-storage 3-2.3:1.2
[38455.686647] input: Espressif Saola 1 w/WROVER Keyboard as /devices/pci0000:00/0000:00:08.1/0000:08:00.3/usb3/3-2/3-2.3/3-2.3:1.3/0003:239A:80A6.000D/input/input45
[38455.747093] input: Espressif Saola 1 w/WROVER Mouse as /devices/pci0000:00/0000:00:08.1/0000:08:00.3/usb3/3-2/3-2.3/3-2.3:1.3/0003:239A:80A6.000D/input/input46
[38455.747172] input: Espressif Saola 1 w/WROVER Consumer Control as /devices/pci0000:00/0000:00:08.1/0000:08:00.3/usb3/3-2/3-2.3/3-2.3:1.3/0003:239A:80A6.000D/input/input47
[38455.747205] input: Espressif Saola 1 w/WROVER as /devices/pci0000:00/0000:00:08.1/0000:08:00.3/usb3/3-2/3-2.3/3-2.3:1.3/0003:239A:80A6.000D/input/input48
[38455.747257] hid-generic 0003:239A:80A6.000D: input,hidraw6: USB HID v1.11 Keyboard [Espressif Saola 1 w/WROVER] on usb-0000:08:00.3-2.3/input3
[38456.684775] scsi host12: scsi scan: INQUIRY result too short (5), using 36
[38456.684778] scsi 12:0:0:0: Direct-Access     Espressi Saola 1 w/WROVER 1.0  PQ: 0 ANSI: 2
[38456.684929] sd 12:0:0:0: Attached scsi generic sg2 type 0
[38456.685339] sd 12:0:0:0: [sdd] 4097 512-byte logical blocks: (2.10 MB/2.00 MiB)
[38456.685552] sd 12:0:0:0: [sdd] Write Protect is off
[38456.685553] sd 12:0:0:0: [sdd] Mode Sense: 03 00 00 00
[38456.685691] sd 12:0:0:0: [sdd] No Caching mode page found
[38456.685692] sd 12:0:0:0: [sdd] Assuming drive cache: write through
[38456.691181]  sdd: sdd1
[38456.692258] sd 12:0:0:0: [sdd] Attached SCSI removable disk
[38457.164359] FAT-fs (sdd1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[38464.343023] FAT-fs (sdc1): Directory bread(block 8) failed
[38464.343026] FAT-fs (sdc1): Directory bread(block 9) failed
[38464.343027] FAT-fs (sdc1): Directory bread(block 10) failed
[38464.343028] FAT-fs (sdc1): Directory bread(block 11) failed
[38464.343029] FAT-fs (sdc1): Directory bread(block 12) failed
[38464.343030] FAT-fs (sdc1): Directory bread(block 13) failed
[38464.343031] FAT-fs (sdc1): Directory bread(block 14) failed
[38464.343032] FAT-fs (sdc1): Directory bread(block 15) failed
[38464.343034] FAT-fs (sdc1): Directory bread(block 16) failed
[38464.343035] FAT-fs (sdc1): Directory bread(block 17) failed

jerryneedell, 2020-07-09:
here is what I see with dmesg (Ubuntu 20.04)
```
[82416.830309] usb 3-1.2.1: new full-speed USB device number 10 using
xhci_hcd
[82417.052646] usb 3-1.2.1: New USB device found, idVendor=239a,
idProduct=80a6, bcdDevice= 1.00
[82417.052651] usb 3-1.2.1: New USB device strings: Mfr=2, Product=3,
SerialNumber=1
[82417.052655] usb 3-1.2.1: Product: Saola 1 w/WROVER
[82417.052657] usb 3-1.2.1: Manufacturer: Espressif
[82417.052659] usb 3-1.2.1: SerialNumber: 0000000000000000160000000CFCDF
[82417.072009] usb-storage 3-1.2.1:1.2: USB Mass Storage device detected
[82417.072557] scsi host7: usb-storage 3-1.2.1:1.2
[82417.075370] input: Espressif Saola 1 w/WROVER Keyboard as
/devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1.2/3-1.2.1/3-1.2.1:1.3/0003:239A:80A6.000A/input/input29
[82417.135076] input: Espressif Saola 1 w/WROVER Mouse as
/devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1.2/3-1.2.1/3-1.2.1:1.3/0003:239A:80A6.000A/input/input30
[82417.135628] input: Espressif Saola 1 w/WROVER Consumer Control as
/devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1.2/3-1.2.1/3-1.2.1:1.3/0003:239A:80A6.000A/input/input31
[82417.135924] input: Espressif Saola 1 w/WROVER as
/devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1.2/3-1.2.1/3-1.2.1:1.3/0003:239A:80A6.000A/input/input32
[82417.136425] hid-generic 0003:239A:80A6.000A: input,hidraw1: USB HID
v1.11 Keyboard [Espressif Saola 1 w/WROVER] on usb-0000:00:14.0-1.2.1/input3
[82417.169865] cdc_acm 3-1.2.1:1.0: ttyACM0: USB ACM device
[82417.170748] usbcore: registered new interface driver cdc_acm
[82417.170749] cdc_acm: USB Abstract Control Model driver for USB modems
and ISDN adapters
[82418.083178] scsi host7: scsi scan: INQUIRY result too short (5), using 36
[82418.083188] scsi 7:0:0:0: Direct-Access     Espressi Saola 1 w/WROVER
1.0  PQ: 0 ANSI: 2
[82418.083904] sd 7:0:0:0: Attached scsi generic sg2 type 0
[82418.084508] sd 7:0:0:0: [sdc] 4097 512-byte logical blocks: (2.10
MB/2.00 MiB)
[82418.084719] sd 7:0:0:0: [sdc] Write Protect is off
[82418.084726] sd 7:0:0:0: [sdc] Mode Sense: 03 00 00 00
[82418.084916] sd 7:0:0:0: [sdc] No Caching mode page found
[82418.084925] sd 7:0:0:0: [sdc] Assuming drive cache: write through
[82418.131028]  sdc: sdc1
[82418.133264] sd 7:0:0:0: [sdc] Attached SCSI removable disk
[82418.702998] FAT-fs (sdc1): Volume was not properly unmounted. Some data
may be corrupt. Please run fsck.
jerryneedell@jerryneedell-ubuntu-macmini:~$
```

tannewt, 2020-07-09:
There have been some bug fixes in TinyUSB I believe. I'll try updating it and we can see if it fixes the issues.

PR #3262 (2020-08-12): Fix writing sector 0 as the first write.
This was the FS issue I saw when debugging wifi and only happens
when the first write is to sector 0. It causes issues because it
blanks all of sector 0 after the first block.

Fixes #3133
commit 87c78be8 (2020-08-12): Fix writing sector 0 as the first write.

This was the FS issue I saw when debugging wifi and only happens
when the first write is to sector 0. It causes issues because it
blanks all of sector 0 after the first block.

Fixes #3133
closed 2020-08-12 by tannewt