adafruit/circuitpython #3139: Crash into safe mode from REPL due to voltage dip when connecting I2C breakout

siddacious, 2020-07-09:
Board is a Metro M4 with UF2 2.0.0
Sensor is a prototype AS7341 but it will occur with nearly every sensor I use

[image]
I will update my UF2 version and try to get some info on the voltage levels

dhalbert, 2020-07-09:
I have reproduced this on CircuitPython 5.3.0 on a Metro M0 and on A Feather M4. The Feather seems a bit less sensitive: a couple of sensors did not trigger this. I did not reproduce it on a Feather Sense (nRF52840).

It happens when plugging in a STEMMA sensor, or if you interrupt the power lines on one that is already plugged in.

dhalbert, 2020-07-09:
I am not sure if there's anything to fix or work around on the software side. If there really is a glitch in the 3.3V rail that drops it below the 2.7V brownout level, then that's an electrical issue which could be significant. For instance, the external QSPI flash chips are not spec'd for operation below 2.7V, and could read or write bad data if their voltage dips.

siddacious, 2020-07-10:
@dhalbert Indeed there is a pretty substantial dip, well below 2.7V:
[image]

Here's the capture from the screenshot above:
https://www.dropbox.com/s/gk8usbzqbtujw2x/sensor_insert_vcc_sag.sal?dl=0
Note that this is from the Logic 2 alpha so you'll have to have that installed to view it

I'll close this since it does seem to be unrelated to CircuitPython

closed 2020-07-10 by siddacious