adafruit/circuitpython #3150: USB_HID report_id = 0 for output reports

CatalinPopovici, 2020-07-14:
Hi,
	
	I need to use usb_hid for a Braille Display. Currently I am using a custom 
	USB Descriptor and every time I am sending an OUT Report to the Itsybitsy M4 Express,
	the usb_hid device implementation calls the wrong callback function. By sending an OUT Report, instead of entering
	the "hidd_control_complete"(circuitpython/lib/tinyusb/src/class/hid/hid_device.c)
	function, it enters the "hidd_xfer_cb"(circuitpython/lib/tinyusb/src/class/hid/hid_device.c)
	function which calls "tud_hid_set_report_cb"(circuitpython/shared-module/usb_hid/Device.c)
	with the report_id=0. By calling the wrong function and sending the report_id forward I 
	can't implement my usb_hid OUT Report functionalities.

	Best regards,
	Catalin Popovici!

dhalbert, 2020-07-14:
I think this might be the same issue as #2348. We haven't tried to implement host-to-device reports yet, so anything that's there may be wrong, as I think you've discovered.

tannewt, 2020-07-14:
Hi! Two questions:
1) What Braille Display are you using? I'd love to see how to set it up and possibly get one to test with.
2) What is your current CircuitPython code? It'd be helpful to see the highest level of code.
Thanks!

tannewt, 2020-07-16:
Is the HID definition for braille display public? All I can find is: https://blogs.microsoft.com/on-the-issues/2018/05/31/technology-sector-collaborates-to-develop-hid-standard-for-braille-displays/

I think that would help us understand what needs to be supported.

dhalbert, 2021-12-24:
We now support custom HID devices, with or without explicit report IDs. I think that closes this.
https://learn.adafruit.com/custom-hid-devices-in-circuitpython

closed 2021-12-24 by dhalbert