adafruit/circuitpython #3151: Offer control over channels used for BLE advertising

kevinjwalters, 2020-07-14:
Bluetooth Low Energy advertises over one of three channels by design. This design is intended to reduce the chance of congestion or jamming breaking the advertising mechanism. In general, this makes a lot of sense in an uncontrolled environment but for some uses this isn't optimal as the expected number of received packets is 1/3 of transmissions.

1. For exchanging data within a room which might be a classroom, I suspect the single channel approach is superior in terms of low-loss communication and jamming would rarely be an issue and could be ameliorated by configuration for the rare cases.
2. `broadcast()` in https://github.com/adafruit/Adafruit_CircuitPython_BLE_BroadcastNet look like it currently fires off a single (possibly fragmented) message with its defaults. Each broadcast is therefore guaranteed not to be heard by 2/3 of the recipients.
3. The investigations into the gargantuan challenge of using tx power and rssi to obtain a distance approximation might benefit a little from this, cf #2826 
4. A niche case is aiding understanding of what's going on in [BLE Advertisement strange, acute packet loss](https://forums.adafruit.com/viewtopic.php?f=60&t=166440).

A relevant comparison here is the basic radio functionality offered for the BBC micro:bit as seen in https://microbit-micropython.readthedocs.io/en/latest/radio.html which has an interesting, concise example by @ntoll. A few years back the micro:bit appeared rather limited to me with BLE unavailable form MicroPython but the radio use got mentioned at a London MeetUp event and it sounded very intriguing. 

Nordic offer control over this with a mask mentioned in https://devzone.nordicsemi.com/f/nordic-q-a/38193/how-to-test-one-channel-advertising - @dhalbert has mentioned this somewhere in the past.

I thought I'd put this ticket in to gauge demand and gather feedback and use cases from others on this feature.

dhalbert, 2020-07-14:
> Bluetooth Low Energy advertises over one of three channels by design. This design is intended to reduce the chance of congestion or jamming breaking the advertising mechanism. In general, this makes [...]

This is not my understanding, and I re-checked with some websearching before replying. By default, a BLE advertisement is transmitted on all three channels 37, 38, 39 for each advertising event. Since the radio can't transmit on multiple frequencies simultaneously, this is done in quick succession. It is possible to turn off one or two of these channels if you want.

Here's a nice picture (from [here](https://microchipdeveloper.com/wireless:ble-link-layer-discovery)) showing what happens. The vertical bars are the advertising transmissions:

[image]

The BroadcastNet used advertising because it's simple and it doesn't matter if it's lossy, since it's reporting sensor readings, where a missed reading doesn't really matter. If you are trying to exchange data that is not going to repeated several times, then I'm not sure advertising is the best idea. Are you trying to do n-to-n conversation as opposed to single peer-to-peer? Maybe sending the data multiple times with a sequence number that increments only when the data changes, so that the receiver knows when something has been dropped, would be helpful. It sounds like you might be reinventing IP multicast

kevinjwalters, 2020-07-14:
@dhalbert That's interesting, I wasn't aware of that. I need to double check the numbers in https://forums.adafruit.com/viewtopic.php?f=60&t=166440 as I thought they gave the appearance of transmission on one channel but perhaps not. It also changes the interpretation of the 5 second steps in https://forums.adafruit.com/viewtopic.php?f=53&t=165613 to be the receiver changing channel every 5 seconds.

For my current application I am just exchanging tiny amounts of data between N devices where N is currently between 2-6. I thought it would an interesting example to explore using advertising packets.

kevinjwalters, 2020-07-14:
The _peak_ numbers in https://forums.adafruit.com/download/file.php?id=75784 do support an all channel advertising from the (pair of) nRF52840. In 30 seconds at 20ms _adding_ 0-10ms jitter each would generate `1/((0.020+0.020+0.010)/2) * 30 = 1200` packets and the rx from the high rx trials are just below that.

I'll close this, thanks for the correction.

closed 2020-07-14 by kevinjwalters

kevinjwalters, 2020-08-03:
Depending on exactly what the requirements are for the testing mentioned in [Adafruit Forums: setting bluetooth channel for CE testing](https://forums.adafruit.com/viewtopic.php?f=53&t=167670), this might be useful there for channel 37 + 39 advertising only.

kevinjwalters, 2020-08-13:
FYI, Nordic have an interesting diagram showing how radio tx/rx works for advertising, channels aren't labelled but there are three of them: https://infocenter.nordicsemi.com/index.jsp?topic=%2Fsds_s140%2FSDS%2Fs1xx%2Fble_power_profiles%2Fadvertising_event.html&cp=4_7_4_0_17_0