adafruit/circuitpython #3153: Auto-reloads based on file modification are deferred if input statement is running

kevinjwalters, 2020-07-14:
The normal mechanism where the program reloads on CircuitPython if a new file is copied across to `CIRCUITPY` appears not to work if the current program is running and is in an `input` statement, i.e. pending user input. When the user presses return the reload does occur so it's blocked rather than lost.

Observed on CircuitPython on 5.3.0.

nk521, 2020-09-08:
Is this bug fixed in 6.0.0-alpha.3-91?

(I tried it on circuitpython playground express board)

jepler, 2020-09-08:
#3318 may have resolved this problem, but it wasn't linked to this issue and thus was not automatically closed.  It is included in 6.0.0-alpha.3 -- If you're confirming that it's fixed, then that's great and I'll go ahead and close this issue.  If not, we'll need to re-test and figure out why the fix was incomplete.

nk521, 2020-09-08:
Well its working fine for me. Lets wait for @kevinjwalters' comment

jepler, 2020-12-30:
I'm closing the issue based on the above comments.  Please feel free to re-open or file a fresh bug report if necessary.

closed 2020-12-30 by jepler

kevinjwalters, 2020-12-30:
Works ok on 6.0.0 on Feather nRF52840 Express.