adafruit/circuitpython #3154: esp32s2: parallel build is broken

jepler, 2020-07-15:
Invoking a "parallel build" in ports/esp32s2 fails with a variety of errors, probably because some elements of the build process are not designed to be run in parallel.  This is unfortunate, because parallel building can greatly decrease build times especially on high performance desktop computers.

For instance, the first overt failure shown:
```
$ make BOARD=espressif_saola_1_wroom clean; make -j19 -O BOARD=espressif_saola_1_wroom
...
[29/30] Linking CXX static library esp-idf/driver/libdriver.a
FAILED: esp-idf/driver/libdriver.a 
: && /home/jepler/.espressif/tools/cmake/3.16.4/bin/cmake -E remove esp-idf/driver/libdriver.a && /home/jepler/.espressif/tools/xtensa-esp32s2-elf/esp-2020r1-8.2.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-ar qc esp-idf/driver/libdriver.a  esp-idf/driver/CMakeFiles/__idf_driver.dir/adc_common.c.obj esp-idf/driver/CMakeFiles/__idf_driver.dir/can.c.obj esp-idf/driver/CMakeFiles/__idf_driver.dir/dac.c.obj esp-idf/driver/CMakeFiles/__idf_driver.dir/gpio.c.obj esp-idf/driver/CMakeFiles/__idf_driver.dir/i2c.c.obj esp-idf/driver/CMakeFiles/__idf_driver.dir/i2s.c.obj esp-idf/driver/CMakeFiles/__idf_driver.dir/ledc.c.obj esp-idf/driver/CMakeFiles/__idf_driver.dir/pcnt.c.obj esp-idf/driver/CMakeFiles/__idf_driver.dir/periph_ctrl.c.obj esp-idf/driver/CMakeFiles/__idf_driver.dir/rmt.c.obj esp-idf/driver/CMakeFiles/__idf_driver.dir/rtc_io.c.obj esp-idf/driver/CMakeFiles/__idf_driver.dir/rtc_module.c.obj esp-idf/driver/CMakeFiles/__idf_driver.dir/sdspi_crc.c.obj esp-idf/driver/CMakeFiles/__idf_driver.dir/sdspi_host.c.obj esp-idf/driver/CMakeFiles/__idf_driver.dir/sdspi_transaction.c.obj esp-idf/driver/CMakeFiles/__idf_driver.dir/sigmadelta.c.obj esp-idf/driver/CMakeFiles/__idf_driver.dir/spi_common.c.obj esp-idf/driver/CMakeFiles/__idf_driver.dir/spi_master.c.obj esp-idf/driver/CMakeFiles/__idf_driver.dir/spi_slave.c.obj esp-idf/driver/CMakeFiles/__idf_driver.dir/spi_bus_lock.c.obj esp-idf/driver/CMakeFiles/__idf_driver.dir/timer.c.obj esp-idf/driver/CMakeFiles/__idf_driver.dir/touch_sensor_common.c.obj esp-idf/driver/CMakeFiles/__idf_driver.dir/uart.c.obj esp-idf/driver/CMakeFiles/__idf_driver.dir/esp32s2/rtc_tempsensor.c.obj esp-idf/driver/CMakeFiles/__idf_driver.dir/esp32s2/touch_sensor.c.obj esp-idf/driver/CMakeFiles/__idf_driver.dir/esp32s2/adc.c.obj esp-idf/driver/CMakeFiles/__idf_driver.dir/esp32s2/adc2_init_cal.c.obj && /home/jepler/.espressif/tools/xtensa-esp32s2-elf/esp-2020r1-8.2.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-ranlib esp-idf/driver/libdriver.a && :
/home/jepler/.espressif/tools/xtensa-esp32s2-elf/esp-2020r1-8.2.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-ar: esp-idf/driver/CMakeFiles/__idf_driver.dir/esp32s2/adc2_init_cal.c.obj: No such file or directory
ninja: build stopped: subcommand failed.
```

A message also says
```
ninja -C build-espressif_saola_1_wroom/esp-idf esp-idf/esp_ringbuf/libesp_ringbuf.a
ninja: Entering directory `build-espressif_saola_1_wroom/esp-idf'
ninja: warning: premature end of file; recovering
```
which could be relevant, but just what file the problem is in, is unclear.

tannewt, 2020-07-15:
This is because we call out to ninja for dependencies. If we switch CP to ninja we can do it natively instead. The IDF deps also don't rebuild because we don't have their deps in our `make` files.

jepler, 2020-07-15:
In the meantime, adding to the bottom of the Makefile, this prevents build problems by disabling parallel jobs inside make even if requested:
```
.NOTPARALLEL:
```
I can prepare a PR if that's the best fix in the short term.

PR #3155 (2020-07-16): esp32s2: Take care to invoke the sub-build-system only once
This speeds up my build by about a factor of 3, when using an appropriate "-j" flag:

| parallelism flag | before | after |
|---|---|---|
| -j1 | 39.6s | 37.7s |
| -j19 | error | 11.6s |

A "make" which does not need to do anything now takes about 800ms; I did not compare to "before".

(test system is AMD Ryzen 7 3700X with 16 threads)

Closes: #3154
closed 2020-07-16 by tannewt