adafruit/circuitpython #3159: Why does CircuitPython not support characters like ░, ▒, and ▓?

UnsignedArduino, 2020-07-17:
Hello world,

I'm making a file reader, and I have a cursor implemented so far (because in the _far_ future I would like to implement writing someone). I would like to use the character ▓ as my blinking cursor, similar to `nano` on linux.

Unfortunately using these characters results in an `OverflowError`. Is it possible to use these types of characters in CircuitPython?

(I think they are called `extended characters`, because they go from 128 to 255, unlike regular characters that range from 32 to 127)

dhalbert, 2020-07-17:
Could you post the code that raises this error?

UnsignedArduino, 2020-07-17:
Sorry for the late response, I just got the computer on. (finally)

Here is the code:
```python3
# https://learn.adafruit.com/circuitpython-stage-game-library/bouncing-balls
# https://circuitpython-stage.readthedocs.io/en/latest/README.html

import stage
import ugame
import time

def update_text(lines_to_skip=0, highlight=False, highlight_x=0, highlight_y=0):
    terminal.clear()
    with open("bible.txt", mode="rt") as file:
        for _ in range(lines_to_skip):
            for nbr in range(20):
                char = file.read(1)
                if char == "\n":
                    break
        x, y = 0, 0
        while True:
            char = file.read(1)
            if char == "\n":
                y += 1
                x = 0
            else:
                if not char == "\r":
                    if x == highlight_x and y == highlight_y and highlight:
                        terminal.char(x, y, "▓")
                    else:
                        terminal.char(x, y, char)
                x += 1
                if x > 19:
                    y += 1
                    x = 0
                if y > 15:
                    break
            # print(char, end="")

# Screen can fit 20x16 characters
terminal = stage.Text(20, 16)

game = stage.Stage(ugame.display, 10)
game.layers = [
    terminal
]
game.render_block()

lines_to_skip = 0
cursor_x = 0
cursor_y = 0
highlight = False
last_highlighted = 0
changed = True

while True:
    buttons_pressed = ugame.buttons.get_pressed()
    if buttons_pressed & ugame.K_UP and cursor_y > 0:
        cursor_y -= 1
        changed = True
    if buttons_pressed & ugame.K_DOWN:
        cursor_y += 1
        changed = True
    if buttons_pressed & ugame.K_LEFT:
        if cursor_x > 0 and cursor_y > 0:
            cursor_x -= 1
        else:
            cursor_x = 19
            cursor_y -= 1
        changed = True
    if buttons_pressed & ugame.K_RIGHT:
        if cursor_x < 19:
            cursor_x += 1
        else:
            cursor_x = 0
            cursor_y += 1
        changed = True
    if time.monotonic() - last_highlighted >= 0.5:
        highlight = not highlight
        last_highlighted = time.monotonic()
        changed = True
    if changed:
        update_text(lines_to_skip, highlight, cursor_x, cursor_y)
        game.render_block()
        changed = False
    game.tick()
```

Error:
```
Traceback (most recent call last):
  File "code.py", line 79, in <module>
  File "code.py", line 33, in update_text
  File "code.py", line 25, in update_text
  File "stage.py", line 558, in char
OverflowError: value must fit in 1 byte(s)
```

UnsignedArduino, 2020-07-17:
Oh and to type ▓, on a Windows computer, pressing `ALT` and `178` (`177` for ▒ and `176` for ░) and releasing `ALT` should type it. Typing it in `code.py` with Mu shows the character correctly but in the serial console (on Mu), it shows up as `â`

tannewt, 2020-07-17:
This looks like a stage library issue. @deshipu may have ideas.

Note that we don't support extended ascii, Python3 and CircuitPython use UTF-8 for it's native string encoding. The "dark shade" character is 3 bytes long in UTF-8.

Mu serial support is a known issue: https://github.com/mu-editor/mu/issues/797

I've seen a difference in iTerm vs Terminal on Mac as well.

UnsignedArduino, 2020-07-17:
Typing `ALT` and `178` in PuTTY gets me " ²" (in PuTTY, on the PyGamer screen, I see "2"), even though I set the encoding to UTF-8.

UnsignedArduino, 2020-07-17:
> This looks like a stage library issue.

Should I close and create a new thread in the [CircuitPython Stage repository](https://github.com/python-ugame/circuitpython-stage/issues?q=is%3Aissue+is%3Aclosed) mentioning this issue?

deshipu, 2020-07-17:
@UnsignedArduino Stage doesn't use CircuitPython's fonts, it instead has its own pixel font, with more colors. But due to size constraints, it only handles the 128 ASCII characters, and doesn't have any extended or unicode characters.

However, I included a few extra symbols in the font in place of the control characters — you can look at the whole font here: https://raw.githubusercontent.com/python-ugame/circuitpython-stage/master/font/font.bmp

So ``terminal.char(x, y, "\x19")`` might do what you want, for example.

UnsignedArduino, 2020-07-18:
Ahh, okay! Thanks a lot for helping me. 

It seems like that I can specify a font, according to the docs: https://circuitpython-stage.readthedocs.io/en/latest/README.html#stage.Text

> Stage doesn't use CircuitPython's fonts, it instead has its own pixel font, with more colors. But due to size constraints, it only handles the 128 ASCII characters, and doesn't have any extended or unicode characters.

Could I specify a font that includes extended and unicode characters, and `Stage` would use them? If I can, how?

Sorry if I'm bothering you.

deshipu, 2020-07-18:
> Could I specify a font that includes extended and unicode characters, and Stage would use them? If I can, how?

You can do it, but it still can only have 128 characters at the maximum — so you could replace the first 32 symbols with your own, but that's it. To generate the font data, you can use the ``genfont.py`` script from the directory I linked to. You basically edit the ``font.bmp`` file to fit your needs, and then copy the output of the script into your program as the font data.

There is one more thing you could do — use displayio for your application, instead of stage. Stage is really limited, it was designed specifically for simple games, and probably is not the best way to make a book reader. Displayio is much more flexible and can use standard fonts, without limitations to which characters are available.

UnsignedArduino, 2020-07-18:
> use displayio for your application

Ahh, I didn't like how displayio managed text (If you had 30 characters at the beginning and assign a 32 character string, you get an exception and I think the strings can't be more than 1 line tall) and stage seemed like a good replacement for it, since handling text was so intuitive. 

But I've heard that the display_text module got some improvements - should probably go check it out (again)

I'll close this, because I got my thing to work, (yay, now you can type words. But you would go insane if you typed more than a sentence) and it solved my problem. Thank you so much @dhalbert and @tannewt and @deshipu and @jepler!

closed 2020-07-18 by UnsignedArduino