adafruit/circuitpython #3167: STM32 heap error w/Airlift guide

hukuzatuna, 2020-07-19:
I'm running latest CircuitPython 5 (as of yesterday's build) with current libraries. When I try to run the second example in the Airlift Featherwing guide (code included below) I get this error:
```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

You are in safe mode: something unanticipated happened.
The CircuitPython heap was corrupted because the stack was too small.
Please increase the stack size if you know how, or if not:
Please file an issue with the contents of your CIRCUITPY drive at 
https://github.com/adafruit/circuitpython/issues

Press any key to enter the REPL. Use CTRL-D to reload.
```
I'm running the latest code, and have only the required libraries on the STM32:
```
pi@sequoia:~/src/accelerometer $ ls /media/pi/CIRCUITPY/lib/
adafruit_bus_device  adafruit_lsm6ds.mpy  adafruit_requests.mpy  neopixel.mpy
adafruit_esp32spi    adafruit_register    adafruit_seesaw
pi@sequoia:~/src/accelerometer $ 
```
Finally, here's the example code I'm using:
```
# SPDX-FileCopyrightText: 2019 ladyada for Adafruit Industries
# SPDX-License-Identifier: MIT

import board
import busio
from digitalio import DigitalInOut
import adafruit_esp32spi.adafruit_esp32spi_socket as socket
from adafruit_esp32spi import adafruit_esp32spi
import adafruit_requests as requests

# Get wifi details and more from a secrets.py file
try:
    from secrets import secrets
except ImportError:
    print("WiFi secrets are kept in secrets.py, please add them there!")
    raise

print("ESP32 SPI webclient test")

TEXT_URL = "http://wifitest.adafruit.com/testwifi/index.html"
JSON_URL = "http://api.coindesk.com/v1/bpi/currentprice/USD.json"

# If you have an AirLift Featherwing or ItsyBitsy Airlift:
esp32_cs = DigitalInOut(board.D13)
esp32_ready = DigitalInOut(board.D11)
esp32_reset = DigitalInOut(board.D12)

spi = busio.SPI(board.SCK, board.MOSI, board.MISO)
esp = adafruit_esp32spi.ESP_SPIcontrol(spi, esp32_cs, esp32_ready, esp32_reset)

requests.set_socket(socket, esp)

if esp.status == adafruit_esp32spi.WL_IDLE_STATUS:
    print("ESP32 found and in idle mode")
print("Firmware vers.", esp.firmware_version)
print("MAC addr:", [hex(i) for i in esp.MAC_address])

for ap in esp.scan_networks():
    print("\t%s\t\tRSSI: %d" % (str(ap["ssid"], "utf-8"), ap["rssi"]))

print("Connecting to AP...")
while not esp.is_connected:
    try:
        esp.connect_AP(secrets["ssid"], secrets["password"])
    except RuntimeError as e:
        print("could not connect to AP, retrying: ", e)
        continue
print("Connected to", str(esp.ssid, "utf-8"), "\tRSSI:", esp.rssi)
print("My IP address is", esp.pretty_ip(esp.ip_address))
print(
    "IP lookup adafruit.com: %s" % esp.pretty_ip(esp.get_host_by_name("adafruit.com"))
)
print("Ping google.com: %d ms" % esp.ping("google.com"))

# esp._debug = True
print("Fetching text from", TEXT_URL)
r = requests.get(TEXT_URL)
print("-" * 40)
print(r.text)
print("-" * 40)
r.close()

print()
print("Fetching json from", JSON_URL)
r = requests.get(JSON_URL)
print("-" * 40)
print(r.json())
print("-" * 40)
r.close()

print("Done!")
```

hierophect, 2020-07-20:
Could you post the exact version of 5.x you are using? Most stack issues on the STM32 have been caused by an issue fixed in #2866, which was included in the 5.4 release. Can you confirm you are using 5.4 or higher?

hukuzatuna, 2020-07-20:
> Could you post the exact version of 5.x you are using? Most stack issues on the STM32 have been caused by an issue fixed in #2866, which was included in the 5.4 release. Can you confirm you are using 5.4 or higher?

Absolutely! Here's what I'm running:
```
Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 5.3.1 on 2020-07-13; Feather STM32F405 Express with STM32F405RG
```

hukuzatuna, 2020-07-20:
5.4.0-beta.1 fixes it. Sorry to waste your time!

closed 2020-07-20 by hukuzatuna

hierophect, 2020-07-20:
@hukuzatuna no problem, this one bug has come up a lot, but it's different for everybody!