adafruit/circuitpython #3169: Accessing Polygon.points results in safe mode reboot

FoamyGuy, 2020-07-19:
With the following vectorio Polygon example code:
```python
import board
import displayio
from vectorio import Polygon, VectorShape
import time

# Make the display context
splash = displayio.Group(max_size=10)
board.DISPLAY.show(splash)

rect = Polygon(points=[(0,0), (100,0), (100, 30), (0,30)])
rect_palette = displayio.Palette(2)
rect_palette.make_transparent(0)
rect_palette[1] = 0xFF0000
vector_shape = VectorShape(shape=rect, x=1, y=1, pixel_shader=rect_palette)
splash.append(vector_shape)

rect.points = [(0,0), (50,0), (50, 30), (0,30)]

# if we comment out the print statement then we won't reboot to safe mode
print(rect.points)

while True:
    pass
```

The line `print(rect.points)` is causing the device to reboot into safemode. 

I'm not sure if it's possible or how to access any sort of stack trace in a situation like I don't see any python stacktrace printed in the serial console when this occurs. If there are some steps I can follow to get that and provide it here for more info I am happy to do that if someone can direct me how. 

I see this message in the serial after rebooting:
```
You are in safe mode: something unanticipated happened.
CircuitPython core code crashed hard. Whoops!
Crash into the HardFault_Handler.
```

This was tested on PyPortal:
`Adafruit CircuitPython 6.0.0-alpha.1 on 2020-07-03; Adafruit PyPortal with samd51j20`

WarriorOfWire, 2020-07-19:
Thanks for the report!

I'll see if I can repro with my feather m4 express.

WarriorOfWire, 2020-07-19:
```
from vectorio import Polygon
rect = Polygon(points=[(0,0), (100,0), (100, 30), (0,30)])
print(rect.points)
```

This shorter piece of code suffices to reproduce.  The read property is misbehaving.

WarriorOfWire, 2020-07-19:
https://github.com/adafruit/circuitpython/pull/3172

FoamyGuy, 2020-07-23:
Resolved by #3172

closed 2020-07-23 by FoamyGuy