adafruit/circuitpython #3170: vectorio VectorShape does not move with it's Group

FoamyGuy, 2020-07-19:
After adding a VectorShape to a `displayio.Group` changing the groups x and y positions does not move the VectorShape.

In this example we make a `vectorio.Polygon` and `VectorShape` both using `0,0` for x/y position. The VectorShape is added to a group that has it's x/y position set to `50,50` but the VectorShape is still drawn in the top left corner at `0,0`

```python
import board
import displayio
from vectorio import Polygon, VectorShape
from adafruit_display_shapes.rect import Rect

WIDTH = 100
HEIGHT = 30

splash = displayio.Group(max_size=10)
board.DISPLAY.show(splash)

rect = Polygon(points=[(0,0), (WIDTH,0), (WIDTH, HEIGHT), (0,HEIGHT)])
rect_palette = displayio.Palette(2)
rect_palette.make_transparent(0)
rect_palette[1] = 0x00FF00
vector_shape = VectorShape(shape=rect, x=0, y=0, pixel_shader=rect_palette)

display_shape_rect = Rect(0, 0, 40, 20, fill=0x0000FF)

shape_group = displayio.Group(max_size=3, x=50, y=50)
shape_group.append(vector_shape)
shape_group.append(display_shape_rect)
splash.append(shape_group)

while True:
    pass
```
For comparison I've added another smaller rectangle made with `adafruit_display_shapes` and added it to the same group and it does get drawn at x/y position `50,50`

tannewt, 2020-07-20:
@warriorofwire are you aware of this?

WarriorOfWire, 2020-07-20:
Hmm, I'll look into it.  Thanks for the report!

closed 2020-07-21 by tannewt