adafruit/circuitpython #3173: PyPortal serial connection gets frozen on 2020-07-17 build

FoamyGuy, 2020-07-19:
I downloaded the "absolute newest" build for the PyPortal recently and today I noticed with this build flashed on my device if I connect to the device via serial the connection will freeze within a few seconds. In Mu this makes the app not respond. In Putty it stops sending and receiving any messages but can still be closed just fine. 

This is the link to the firmware: https://adafruit-circuit-python.s3.amazonaws.com/bin/pyportal/en_US/adafruit-circuitpython-pyportal-en_US-20200717-9cdf5e1.uf2

If I download the [20200719 firmware](https://adafruit-circuit-python.s3.amazonaws.com/bin/pyportal/en_US/adafruit-circuitpython-pyportal-en_US-20200719-8f92834.uf2) and flash it this issue does seem to be resolved. I wanted to make a note of it just in case it's unknown or if it could still be a problem. I'm not so sure what the cause of it could be.

FoamyGuy, 2020-07-19:
I tested the 2020-07-17 build for Edge Badge and I am not having a similar issue. This may be something PyPortal specific, or perhaps even a hardware issue on my device. If someone else has a chance to test out that same PyPortal build and is not having a problem we can probably close this.

jepler, 2020-07-20:
Odd, the only changes from 9cdf5e1 to 8f92834 are to pewpew_m4.  9cdf5e1 is the merge of the background-callback stuff.

I loaded 9cdf5e1 on my pyportal and ran a script at the REPL to print:
```
Adafruit CircuitPython 6.0.0-alpha.1-108-g9cdf5e148 on 2020-07-17; Adafruit PyPortal with samd51j20
>>> while True: print("hi ", time.time()) or time.sleep(.2)
hi  946684984
hi  946684985
hi  946684985
...
```
It ran for at least 30 seconds before I stopped it.

Please feel free to re-open this issue if you reproduce it--there could be something else specific about what you were doing in your first test that you didn't the second time.  Additional information almost always helps.

closed 2020-07-20 by jepler