adafruit/circuitpython #3179: Use gcc10 eventually

dhalbert, 2020-07-21:
The arm gcc10 toolchain is available in preview form here: https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads (currently second item in downloads list). This preview was released in late June, 2020, and is not recommended for production use.

MicroPython has had to make some changes to fix compile errors: [gcc 10 optimises memset to call itself
](https://github.com/micropython/micropython/issues/6053).

Compilation speed benchmarks: https://github.com/nordlow/compiler-benchmark

tannewt, 2020-07-22:
I did have to bring in the patch Damien made. It didn't start without it.

tannewt, 2020-12-14:
ARM has released GCC 10.2: https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads

closed 2020-12-17 by dhalbert