adafruit/circuitpython #3181: displayio: core changes to enable decorator and/or context handler to batch display changes

jepler, 2020-07-21:
@foamyguy has written (https://github.com/adafruit/Adafruit_CircuitPython_ProgressBar/pull/8):

> When the progress is moving down the library previously would fill the "empty" portion of the progress bar from left to right so it looked like it was moving "the wrong way" I modified the behavior to fill emptiness from right to left so the progress bar appears to visually "empty" itself in reverse as it gets drawn.

This prompts me to imagine that there's a missing piece: A decorator and/or context-handler that ensures that all the displayio changes done inside it are displayed at the same time, not piecemeal.

Complicating factor:  In the current world, it's actually the `display` that needs its `auto_refresh` flag handled, but a `ProgressBar` instance does not have access to the related `display` object.  I think something needs to be added in the core to help facilitate this.

Alternative: Pursue it as a bug that multiple updates aren't getting batched due to the "target fps" of the display presumably being much larger than the length of time it takes to actually do  "a few hundred" bitmap updates.  (there may be more work than necessary done in ProgressBar, https://github.com/adafruit/Adafruit_CircuitPython_ProgressBar/issues/10)

tannewt, 2020-07-21:
I think this is a bit unique to bitmap modifications because it takes time to mutate it. When a refresh comes in it'll refresh the mutated area of the bitmap. Keeping those changes local will make that rectangle smaller. This also applies to tilegrids that are greater than 1x1 tiles.

jepler, 2020-07-23:
If it's too niche an idea feel free to close this up.

kmatch98, 2021-02-27:
This issue is relevant to animations in widgets for the https://github.com/adafruit/Adafruit_CircuitPython_DisplayIO_Layout, this is primarily to bitmap redraws, often with a `fill followed by a `blit`.  In this process we need to a flash on the screen that will occur if an auto_refresh triggers between the fill and blit. 

I’ve been defaulting to passing in the `display` so that the widget can toggle the `display.auto_refresh` to pause (False) and then batch changes (when set to True). The downside is that your display must be setup prior to the instance of the widget. 

Maybe you are envisioning some property tied to a Group where changes can be batched and then released. Right now a Group can be hidden, but there is no way to delay a refresh. The presence of a “frozen, don’t refresh me” flag would cause complications if you have some overlapping features that could trigger dirty rectangles appearing on the “frozen ” group. 

Rather than flagging specific structures as “frozen”, it would be easier if there was an global flag that could prevent any display refreshes, maybe it could somehow halt the overall auto_refresh timer that is used to check for scheduling screen refreshes. I can’t quite see how refresh gets called and where the timer resides.  Any insights are welcome.