adafruit/circuitpython #3184: Displaying External Bs on the RGB Matrix is resulting in Safe Mode

makermelissa, 2020-07-21:
I'm using a Metro M4 WiFi and the RGB Matrix Shield w/ bodge wire. I wrote a simple script that displays a bitmap using the usual display

makermelissa, 2020-07-21:
I noticed using a small palette bitmap doesn't seem to result in this issue, so maybe it's a memory thing.

makermelissa, 2020-07-22:
This more specifically happens when bit_depth is 3 or higher.

jepler, 2020-07-23:
Can you send me a complete example?  Is it possible this affects only airlift?  Here's the example I cooked up based on your description.  However, I didn't get any crashes (tested with 5.3.1, 6.0.0-alpha.1, and latest on metro m4 *non-airlift*):
```
import board
import displayio
import digitalio
import framebufferio
import rgbmatrix

displayio.release_displays()

for a in [board.A0, board.A1, board.A2, board.A3]:
    d = digitalio.DigitalInOut(a)
    with d:
        d.switch_to_output(True)
    
matrix = rgbmatrix.RGBMatrix(
    width=64,
    height=32,
    bit_depth=5,
    rgb_pins=[board.D2, board.D3, board.D4, board.D5, board.D6, board.D7],
    addr_pins=[board.A0, board.A1, board.A2, board.A3],
    clock_pin=board.A4,
    latch_pin=board.D10,
    output_enable_pin=board.D9,
)

# Wait for the image to load.
display = framebufferio.FramebufferDisplay(matrix)
f = open("/emoji.bmp", "rb")
b1 = displayio.OnDiskBitmap(f)
tg1 = displayio.TileGrid(b1, pixel_shader=displayio.ColorConverter())
g1 = displayio.Group(max_size=3) 
g1.append(tg1)
display.show(g1)
display.refresh()
print("we got to the loop")
while True:
    pass
```
note also that I didn't hook up a display.

Here's a zip of the bmp I used:
[emoji.zip](https://github.com/adafruit/circuitpython/files/4966922/emoji.zip)

makermelissa, 2020-07-23:
Try this and wait like 30 seconds:
```
import board
import displayio
import framebufferio
import rgbmatrix
import terminalio

displayio.release_displays()
matrix = rgbmatrix.RGBMatrix(
    width=64, height=32, bit_depth=4,
    rgb_pins=[board.D2, board.D3, board.D4, board.D5, board.D6, board.D7],
    addr_pins=[board.A0, board.A1, board.A2, board.A3],
    clock_pin=board.A4, latch_pin=board.D10, output_enable_pin=board.D9)
display = framebufferio.FramebufferDisplay(matrix, auto_refresh=False)

splash = displayio.Group(max_size=15)
display.show(splash)
        
bg_sprite = displayio.TileGrid(
    displayio.OnDiskBitmap(open("red.bmp", "rb")),
    pixel_shader=displayio.ColorConverter(),
    x=0,
    y=0,
)

splash.append(bg_sprite)
display.refresh(target_frames_per_second=60)

while True:
    pass
```

Here's the bitmap I used
[red.bmp.zip](https://github.com/adafruit/circuitpython/files/4967353/red.bmp.zip)

This could be related to lighting up the fullscreen or something.

makermelissa, 2020-07-23:
It could be related to the `target_frames_per_second=60`. I removed that from my MatrixPortal library and I can get it working solidly at bit_depth=3, but it still fails on 4.

PaintYourDragon, 2020-08-27:
Unscientific observation: I was getting these panic stops A LOT @ 6bpp… but was also monitoring the serial port and editing code on the board at the same time. Disconnect board so it’s standalone, it generally seems stable even with bit_depth=6. I’ll make noise if that changes.

closed 2020-09-02 by tannewt