adafruit/circuitpython #3201: displayio.Shape does not change after calling set_boundary() again

FoamyGuy, 2020-07-25:
I have this script:
```python
import displayio
import board
import time

WIDTH = 100
HEIGHT = 100
display = board.DISPLAY

shape = displayio.Shape(WIDTH, HEIGHT, mirror_x=False, mirror_y=False)
for y in range(HEIGHT):
    shape.set_boundary(y, 0, 30)

palette = displayio.Palette(2)
palette[0] = 0x0000FF
palette[1] = 0x00FFFF

tile_grid = displayio.TileGrid(shape, pixel_shader=palette)
group = displayio.Group()
group.append(tile_grid)
display.show(group)

#time.sleep(1)
while True:
    for y in range(HEIGHT):
        shape.set_boundary(y, 0, 80)
    time.sleep(1)
    for y in range(HEIGHT):
        shape.set_boundary(y, 0, 30)
    time.sleep(1)
```
If I run this code as-is I will see the 80px wide shape drawn on the screen but it will never change to 30px wide even though are changing the shape with new calls to `set_boundary()` inside the main loop. 

If I un-comment the `time.sleep(1)` on the line before the main loop then I will see the 30px wide shape drawn on the screen, but never the 80px one. 

The shape seems to get "stuck" at a certain size and does not change visually with further changes. Calling `time.sleep()` seems to have some impact on which size the shape is shown at, but it's "stuck" either way, never changing visibly.

I tested this on PyPortal and PyGamer both versions 5.3.1 and 6.0.0.alpha2.

anecdata, 2020-07-25:
Forcing the show by manually doing
```
    display.show(None)
    display.show(group)
```
after each `for` loop gets the shapes toggling.

Still dependent on the sleep value... no sleep toggles blockily. 1 or 0.1 seconds are fine. 0.01 seconds freezes again. Interesting behaviors.

tannewt, 2020-07-27:
We'll need to add dirty tracking to Shape to make this work. Similar to this: https://github.com/adafruit/circuitpython/blob/main/shared-module/displayio/Bitmap.c#L153-L164

kmatch98, 2020-09-03:
I’m starting to look into this issues.  @tannewt, I will appreciate if you can confirm I am heading in the right direction:

shared_module\Group.c has a function called `displayio_group_get_refresh_areas` that seeks out the refresh areas of `Vectorio`, `TileGrid` or `Group` display elements.  Also there is a corresponding function `displayio_group_finish_refresh` that Is called after a refresh is completed, that also has entries for the same three display structures.

I suspect that a new entry should be added for the `Shape` element and the associated `shape_get_refresh_areas`, `shape_finish_refresh` and internal data structure for storing the `Shape` dirty rectangle tracking.

tannewt, 2020-09-03:
I have no idea why sleep impacts it.

Group shouldn't need to change because Shape acts similar to Bitmap. TileGrid manages it's refresh area. You can see one TODO about it here: https://github.com/adafruit/circuitpython/blob/main/shared-module/displayio/TileGrid.c#L502

You'll also want to do something similar to this: https://github.com/adafruit/circuitpython/blob/main/shared-module/displayio/TileGrid.c#L533

kmatch98, 2020-09-04:
@FoamyGuy and @tannewt I replicated your observation about the screen drawing related to `time.sleep` on an NRF52840 Itsy Bitsy.  

Here is one hypothesis on why sometimes the drawing updates.  In your example code, perhaps the Shape boundary gets updated to the 80 size before the screen refresh timing hits.  So, even though there is no dirty rectangle tracking, if the display refresh cycle hits after the screen was drawn at 30 and then at 80, then it will only show the 80 size.  

I fiddled with the delay and actually observed something in-between drawn:
```
display.refresh(target_frames_per_second=60, minimum_frames_per_second=60)
display.show(group)
time.sleep(1/100)
```
### On an NRF52840 ItsyBitsy

[image]

FoamyGuy, 2021-01-23:
fixed by #3374

closed 2021-01-23 by FoamyGuy