adafruit/circuitpython #3204: vectorio.Polygon x position off by one

FoamyGuy, 2020-07-25:
Using this script:
```python
import board
import displayio
from vectorio import Polygon, VectorShape

# Make the display context
splash = displayio.Group(max_size=20)
board.DISPLAY.show(splash)

# Much easier to see white pixels
color_bitmap = displayio.Bitmap(320, 240, 1)
color_palette = displayio.Palette(1)
color_palette[0] = 0xFFFFFF
bg_sprite = displayio.TileGrid(color_bitmap, x=0, y=0, pixel_shader=color_palette)
splash.append(bg_sprite)
##########################################################################
vector_rect = Polygon(points=[(0,0), (50,0), (50, 50), (0,50)])
rect_palette = displayio.Palette(2)
rect_palette.make_transparent(0)
rect_palette[1] = 0xFF0000
vector_shape = VectorShape(shape=vector_rect, x=0, y=0, pixel_shader=rect_palette)
vector_group = displayio.Group(x=0,y=0)
vector_group.append(vector_shape)
splash.append(vector_group)

while True:
    pass
```
I believe the red square should get drawn at pixel coordinates `0,0` on the screen because it's Polygon points, VectorShape, and Group all use coordinates `0,0`. But it is actually ending up drawn at `1,0` with one pixel of space to it's left.

[image]

Tested on PyPortal with CircuitPython 6.0.0-alpha.2

@WarriorOfWire Let me know if there is any other helpful info that I could provide about this.

WarriorOfWire, 2020-07-29:
This will place the rectangle's top-left position as expected and with your code it should provide a 50x50 rect.  It will not draw the X=50 column or the Y=50 row.
https://github.com/adafruit/circuitpython/pull/3221

tannewt, 2020-07-29:
@WarriorOfWire This fixed by #3221 ?

WarriorOfWire, 2020-07-29:
Yes, that's the answer.  😊

closed 2020-07-29 by tannewt