adafruit/circuitpython #3206: I2C failing

jerryneedell, 2020-07-25:
I tried some I2C devices on a fresh build of CP 6.0.0-alpha2  and am getting the following errors
```

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 6.0.0-alpha.2-24-ga6e048686 on 2020-07-25; Adafruit Metro M4 Airlift Lite with samd51j19
>>> 
>>> import ahtx0_simpletest
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "ahtx0_simpletest.py", line 6, in <module>
  File "adafruit_ahtx0.py", line 68, in __init__
  File "adafruit_ahtx0.py", line 78, in reset
  File "adafruit_ahtx0.py", line 78, in reset
  File "adafruit_bus_device/i2c_device.py", line 104, in write
TypeError: extra keyword arguments given
>>> 
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Hello World!

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 6.0.0-alpha.2-24-ga6e048686 on 2020-07-25; Adafruit Metro M4 Airlift Lite with samd51j19
>>> import msa301_simpletest
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "msa301_simpletest.py", line 8, in <module>
  File "adafruit_msa301.py", line 221, in __init__
  File "adafruit_register/i2c_bit.py", line 70, in __set__
  File "adafruit_register/i2c_bit.py", line 70, in __set__
  File "adafruit_bus_device/i2c_device.py", line 104, in write
TypeError: extra keyword arguments given
>>> 

```

dhalbert, 2020-07-25:
This is due to https://github.com/adafruit/Adafruit_CircuitPython_BusDevice/issues/53. We need to update the library not to use the was-deprecated-and-now removed `stop=` keyword argument.

jerryneedell, 2020-07-25:
ouch...

jerryneedell, 2020-07-26:
resolved by #54 and redundant with #53

closed 2020-07-26 by jerryneedell