adafruit/circuitpython #3207: CircuitPython 6.0.0-alpha.2 - Circuit Playground Express + Crickit

RealAbsurdity, 2020-07-25:
This is my first time ever posting an issue to a public project on GitHub, so please bear with me if I break any norms. I'm interested in contributing and learning this process as I go.

I am attempting to use the CPX + Crickit 6.0 alpha 2 release so I can take advantage of the new countio module. The countio module in 6.0.0 a2 works great on its own when I test it.

When I use the same 6.0.0 alpha 2 CPX + Crickit version of CircuitPython and attempt initialize the Crickit and motor I get an error. It appears that it is not able to communicate with the crickit seesaw device with the CPX i2c pins. The error occurs when "from adafruit_crickit import crickit" is called on line 3 of my code.py file.

I'm doing this with the following import which works fine in CircuitPython 5.3.1 for CPX + Crickit: 

///

from adafruit_crickit import crickit
import neopixel
import time
import board

///

Serial Output:

code.py output:
Traceback (most recent call last):
  File "code.py", line 3, in <module>
  File "adafruit_crickit.py", line 419, in <module>
  File "adafruit_seesaw/seesaw.py", line 152, in __init__
  File "adafruit_seesaw/seesaw.py", line 156, in sw_reset
  File "adafruit_seesaw/seesaw.py", line 422, in write8
  File "adafruit_seesaw/seesaw.py", line 451, in write
  File "adafruit_seesaw/seesaw.py", line 451, in write
  File "adafruit_bus_device/i2c_device.py", line 104, in write
TypeError: extra keyword arguments given

Press any key to enter the REPL. Use CTRL-D to reload.

jepler, 2020-07-25:
Hi and thanks for giving this pre-release version of CircuitPython a try.  We appreciate your feedback.

It looks like this may be a problem to be resolved in the adafruit_bus_device library, consider subscribing to this issue to keep up on new developments: https://github.com/adafruit/Adafruit_CircuitPython_BusDevice/issues/53 -- there's also another similar issue filed here, https://github.com/adafruit/circuitpython/issues/3206

I suspect that in the next few days an updated version of the CircuitPython bundle will be released that resolves this problem.

tannewt, 2020-07-27:
@RealAbsurdity Thank you for the issue! It should be fixed in the latest bundle with Bus Device version 5.0.1.

closed 2020-07-27 by tannewt

RealAbsurdity, 2020-08-03:
> @RealAbsurdity Thank you for the issue! It should be fixed in the latest bundle with Bus Device version 5.0.1.

Thanks for taking a look at it! Just to clarify, will Bus Device version 5.0.1 will be in a nightly release of the 6.0 pre-release alpha? Or will I need to update the library i'm using for the Crickit?

tannewt, 2020-08-03:
It was updated 12 hours ago so the latest S3 build should have it.

RealAbsurdity, 2020-08-04:
Thank you all! Confirmed working with the August 3 release in S3 and the latest August 4 libraries.