adafruit/circuitpython #3216: possible pulseio issue -- irremote not working

jerryneedell, 2020-07-28:
I have been trying to run the irremote_simpletest on the current master and it is hanging in read_pulses(pulsein)

Not sure if this is an issue for the irremote library  or CP

I tried it on both a CPX and a pirkey_m0

```
Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 6.0.0-alpha.2-58-gdebbf1028 on 2020-07-28; Adafruit CircuitPlayground Express with samd21g18
>>> import irremote_simpletest

```
it works with 5.3.1
```
Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 5.3.1 on 2020-07-13; Adafruit CircuitPlayground Express with samd21g18
>>> 
>>> import irremote_simpletest
Heard 5 Pulses: [8104, 179, 690, 83, 656]
Failed to decode:  ('10 pulses minimum',)
----------------------------
Heard 67 Pulses: [9698, 4416, 627, 548, 621, 552, 596, 552, 657, 520, 599, 574, 624, 522, 626, 549, 620, 555, 593, 1687, 620, 1658, 650, 1659, 618, 1660, 617, 1660, 647, 1662, 595, 1684, 624, 1652, 625, 1684, 593, 554, 626, 1681, 625, 1655, 652, 1628, 620, 556, 596, 577, 628, 520, 678, 499, 620, 1659, 627, 549, 620, 555, 623, 524, 625, 1682, 625, 1656, 592, 1687, 649]
Decoded: [255, 0, 71, 184]
----------------------------

```

code for reference
```
# Circuit Playground Express Demo Code
# Adjust the pulseio 'board.PIN' if using something else
import pulseio
import board
import adafruit_irremote

pulsein = pulseio.PulseIn(board.REMOTEIN, maxlen=120, idle_state=True)
decoder = adafruit_irremote.GenericDecode()

while True:
    pulses = decoder.read_pulses(pulsein)
    print("Heard", len(pulses), "Pulses:", pulses)
    try:
        code = decoder.decode_bits(pulses)
        print("Decoded:", code)
    except adafruit_irremote.IRNECRepeatException:  # unusual short code!
        print("NEC repeat!")
    except adafruit_irremote.IRDecodeException as e:  # failed to decode
        print("Failed to decode: ", e.args)

    print("----------------------------")
```

jerryneedell, 2020-07-28:
FYI -- the irremote_simpletest works normally on a feather_nrf52840  (bluefruit_sense)  using an eternal IR sensor.
```
Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 6.0.0-alpha.2-60-g945736e9b on 2020-07-28; Adafruit Feather Bluefruit Sense with nRF52840
>>> import irtest
Heard 67 Pulses: [9064, 4422, 618, 534, 622, 531, 624, 528, 617, 536, 620, 533, 622, 531, 615, 537, 618, 1604, 618, 1603, 618, 1604, 617, 1604, 618, 1604, 617, 1605, 616, 1606, 616, 1607, 624, 529, 616, 538, 618, 535, 620, 534, 622, 1600, 621, 533, 623, 530, 615, 538, 618, 535, 620, 1602, 619, 1603, 619, 1603, 618, 535, 621, 1600, 621, 1601, 620, 1602, 620, 1602, 619]
Decoded: [254, 1, 239, 16]
----------------------------
```

tannewt, 2020-07-28:
@daveputz any thoughts?

DavePutz, 2020-07-28:
We did test the fix for issue #3049 on a CPX with irremote - I will re-test and investigate further.

DavePutz, 2020-07-28:
Looks like this was broken by commit 9cdf5e1. Connie determined that the
previous build
(commit 72f6d7e )  works but  9cdf5e1 ( Jeff's linked list of background
tasks ) does not.
I will investigate to see exactly what the change affected.

DavePutz, 2020-07-31:
The redo of how background tasks are handled changed the accuracy of supervisor_background_tasks_ok() in the Pulsein interrupt handler. This resulted in the handler pausing the collecting of the pulsein stream, which effectively hangs the application. I will address 2 issues in a PR: 1) Correcting supervisor_background_tasks_ok() so that is is again accurate 2) Proper handling of an actual case of pulsein taking too long, returning an error to the application instead of hanging.

closed 2020-09-16 by jepler