adafruit/circuitpython #3250: Optimization level -O2 causing displayio issues

DavePutz, 2020-08-05:
See issue #3247. The problem was reported on a Clue running the demo spirit level script. According to @dhalbert this kind of behavior has typically meant that something that should be volatile is not, or that there's some uninitialized variable.

DavePutz, 2020-08-14:
So, this issue has been narrowed down to the code in spim_xfer() in ports/nrf/nrfx/drivers/src/nrfx_spim.c
It can be resolved by un-commenting the line
`#define NRFX_SPIM3_NRF52840_ANOMALY_198_WORKAROUND_ENABLED 1`
in nrfx_config.h. However, there is a comment that says un-commenting that line breaks BLE.
 @dhalbert  - do you know if that comment is correct? If it is, what options do we have for building with -O2? I did find that
I could surround spim_xfer() with #pragmas to set -Os optimization for just that function, but the use of those in production is
discouraged...

dhalbert, 2020-08-15:
Yes, that comment is correct. The other workaround for the SPIM3 hardware bug is to give SPIM3 its own reserved 8kB block of RAM. I thought I had done that, but I made a mistake in the RAM layout, and need to fix #3147. When I do that, we can try `-O2` again.

dhalbert, 2020-08-15:
Addendum: I wrote that comment, and also pointed to the explanation:
```
// NOTE: THIS WORKAROUND CAUSES BLE CODE TO CRASH.
// It doesn't work with the SoftDevice.
// See https://devzone.nordicsemi.com/f/nordic-q-a/33982/sdk-15-software-crash-during-spi-session
// Turn on nrfx supported workarounds for errata in Rev1 of nRF52840
#ifdef NRF52840_XXAA
//    #define NRFX_SPIM3_NRF52840_ANOMALY_198_WORKAROUND_ENABLED 1
#endif
```

DavePutz, 2020-08-15:
Closing, fixed by PR #3281. Tested OK with a Clue and Circuitplayground Bluefruit with attached Gizmo display.

closed 2020-08-15 by DavePutz

dhalbert, 2020-08-15:
@DavePutz Thanks for testing!

PR #3281 (2020-08-17): SPIM3 buffer must be in first 64kB of RAM
Fixes #3247 and fixes #3250 (@DavePutz, please test, thanks)
Fixes #3147 
Fixes #2946 

#3250: I believe this error is caused by #3147, which notes that the SPIM3 buffer is misplaced. To put it in its own 8kB region, the SoftDevice RAM area had to shrink from 64kB to 56kB. This was done by shrinking `BLEIO_HVN_TX_QUEUE_SIZE` from 9 to 5. The number of connections did not need to be reduced from its present 5.

For nRF52833, the special SPIM3 buffer is not necessary, because the SPIM3 peripheral does not suffer from the same hardware bug as on the nRF52 [... cut]