adafruit/circuitpython #3253: USB VID/PID Request for PicoPlanet

bleeptrack, 2020-08-07:
Hi,
I'm currently working on a series of SAMD21 boards that have procedurally generated decorations and capacitive touch buttons:
https://twitter.com/Bleeptrack/status/1291013322062659586

I'd like to compile my own version of the adafruit trinket bootloader to give CPY support to my boards. Would it be possible to receive a VID/PID for this purpose? This would be amazing.

Thank you very much!

ladyada, 2020-08-07:
those PCBs are so cute!!!!

jepler, 2020-08-07:
I agree! 
[image]

ladyada, 2020-08-07:
```
BleepTrack PicoPlanet    VID 0x239A    PID 0x00C1  # bootloader
                                PID 0x80C1  # arduino
                                PID 0x80C2  # circuitpython
```

closed 2020-08-07 by ladyada

caternuson, 2020-08-07:
> procedurally generated decorations

:astonished: :thumbsup: neat!