adafruit/circuitpython #3259: SDcard f.read() fails on some files >= 1024 size

gmeader, 2020-08-10:
I have a problem reading certain JSON files stored on an SD card on a PyPortal Titano. Basically, if the size of the JSON string in the file is 1024 characters or larger, read() fails. If smaller, it works fine. 
Here’s a short demo Python program. https://github.com/gmeader/pyportal/blob/master/bug_demo.py

anecdata: @gmeader Confirmed. I can read a >=1024 character file from CIRCUITPY, but not from SD Card. https://gist.github.com/anecdata/f59901af2922ad2279971a57894c2a24 (PyPortal 6.0.0-alpha.1 + latest lib)

jepler, 2020-08-10:
Anecdata's traceback:
```
code.py output:
/1023.txt 1023 1023
/sd/1023.txt 1023 1023
/1024.txt 1024 1024
/sd/1024.txt 1024 Traceback (most recent call last):
  File "code.py", line 33, in <module>
OSError: [Errno 5] Input/output error
```

anecdata, 2020-08-18:
@gmeader I wrestled with this for a while, had even reformatted my card at one point during initial testing. Always something odd at 1024 bytes.

I just tried a new card, formatted with the official SD Card format utility, and it worked fine (your example gist code and mine). Odd that we both see errors at around 1024 bytes, which leads me to believe it's not random corruption, but some underlying card characteristic or state that triggers the IO error.

Curious if a new card, newly-formatted, changes the result for you.

FWIW, the original card I tested with was one I've been using without issue in my PyPortal, writing (but also reading) a large log file. It's a Sandisk Ultra 64GB Micro SDXC. The card that worked was a new Sandisk Ultra 16GB Micro SDHC.

DavePutz, 2020-10-19:
I was unable to reproduce this on 6.0.0-rc.0.

tannewt, 2020-10-19:
@gmeader Mind trying rc.0 and seeing if you still have the issue?

gmeader, 2020-10-20:
My demo program now works on 6.0.0-rc.0 on my PyPortal Titano with the latest libraries (6.x-mpy-20201019)

However, multiple Adafruit Learn Guides examples **do not work accessing the Internet**.
Libraries example: pyportal_simpletest.py works fine. esp32spi_simpletest.py also works fine.

These do not work:
1. PyPortal_CMA_ART_FRAME
2. PyPortal_NASA
3. PyPortal_NewNewNew
4. PyPortal_CuteFuzz
5. PyPortal_Tides

Those are all I bothered to try.

The error is usually a wget problem
code.py output:
ESP firmware: bytearray(b'1.6.1\x00')
Connecting to AP MySpectrumWiFi92-2G
Set background to  0
Retrieving data...Socket missing recv_into. Using more memory to be compatible
Reply is OK!
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
ESP firmware: bytearray(b'1.6.1\x00')
Connecting to AP MySpectrumWiFi92-2G
Set background to  0
Retrieving data...Socket missing recv_into. Using more memory to be compatible
Reply is OK!
original URL: https://apod.nasa.gov/apod/image/2010/JupiterSaturnPeaks_Hofer_960.jpg
convert URL: https://io.adafruit.com/api/v2/gmeader/integrations/image-formatter?x-aio-key=eaa01b08cfc437ac1d2ad651e27726a62d64f856&width=320&height=240&output=BMP16&url=https://apod.nasa.gov/apod/image/2010/JupiterSaturnPeaks_Hofer_960.jpg
Fetching stream from https://io.adafruit.com/api/v2/gmeader/integrations/image-formatter?x-aio-key=eaa01b08cfc437ac1d2ad651e27726a62d64f856&width=320&height=240&output=BMP16&url=https://apod.nasa.gov/apod/image/2010/JupiterSaturnPeaks_Hofer_960.jpg
An error occurred, retrying in 60 seconds - wget didn't write a complete file

**Here is output from Pyportal_Tides:**
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
ESP firmware: bytearray(b'1.6.1\x00')
Connecting to AP MySpectrumWiFi92-2G
Set background to  /tides_bg.bmp
Getting time for timezone America/Los_Angeles
Socket missing recv_into. Using more memory to be compatible
struct_time(tm_year=2020, tm_mon=10, tm_mday=19, tm_hour=22, tm_min=37, tm_sec=55, tm_wday=1, tm_yday=293, tm_isdst=None)
Retrieving data...Traceback (most recent call last):
  File "code.py", line 89, in <module>
  File "code.py", line 51, in get_tide_info
  File "adafruit_pyportal.py", line 904, in fetch
  File "adafruit_requests.py", line 632, in get
  File "adafruit_requests.py", line 535, in request
  File "adafruit_requests.py", line 429, in _get_socket
  File "adafruit_requests.py", line 425, in _get_socket
  File "adafruit_requests.py", line 590, in connect
  File "adafruit_esp32spi/adafruit_esp32spi_socket.py", line 75, in connect
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 750, in socket_connect
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 649, in socket_open
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 323, in _send_command_get_response
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 306, in _wait_response_cmd
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 293, in _wait_response_cmd
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 275, in _check_data
RuntimeError: Expected 01 but got 00

Please test the code from these Guides on on 6.0.0-rc.0 on a PyPortal Titano with the latest libraries (6.x-mpy-20201019) as something is definitely wrong.

tannewt, 2020-10-20:
@gmeader I suspect that issue is this: https://github.com/adafruit/Adafruit_CircuitPython_Requests/issues/41

I'm closing this issue because it appears the SDCard issue is fixed.

closed 2020-10-20 by tannewt