adafruit/circuitpython #3264: Refactor PulseOut to not take PWMOut as parameter

hierophect, 2020-08-11:
On the ESP32-S2, PulseIn and PulseOut are handled by a dedicated IR remote control peripheral, the RMT, which can both generate and receive pulse trains with a variable frequency carrier signal. Since this peripheral is completely independent of the ESP32's PWM peripheral, it does not make sense to require a PWMOut object, as is currently required by the API. 

Since there is no set reason why PulseOut needs PWM set up externally, even on other ports, this suggests a change could be made to the PulseOut API that replaces the PWMOut object with parameters for `Pin`, `frequency`, and optionally `duty_cycle` (which will be 50% for most carrier signals). Other ports will need to be modified to call the PWMOut construct API from inside PulseOut construct.

tannewt, 2020-08-11:
I don't think we need to change the API. Internally we can make it work without API changes at the cost of using a timer for the plain PWM.

hierophect, 2020-08-11:
@tannewt how would you like it to appear on the ESP32? Currently, I have it set to change the API based on a mpboardconfig ifdef.

hierophect, 2020-08-11:
Or are you saying that we use a PWM timer and then just de-init it, as I mentioned in Discord?

tannewt, 2020-08-11:
Ya, just have the same API and get the pwm properties from the given object. deiniting after could be weird because it can't be reused then.

hierophect, 2020-08-11:
I'm less of a fan of that because there aren't that many PWM timers and channels available on the ESP32 - this reserves 1 out of only 4 timers without actually using it for anything. If we were to put it in as an API change, I would envision putting it alongside the PWMIO change, since that'll be breaking things anyway.

hierophect, 2020-08-11:
@dhalbert suggested we "expand" the API, but I wasn't quite sure what he meant.

dhalbert, 2020-08-11:
I mean that you could have `PulseOut(pwmout)` and `PulseOut(pin, ...)` coexist, for backward compatibility. The former would be deprecated in favor of the latter. It could be dropped later.

I think having `PulseOut` take a `PWMOut` initially was an unwarranted assumption, based on how `PulseOut` would be implemented on most chips.

hierophect, 2020-08-11:
@dhalbert you have a comparable example in Shared-Bindings? I can kind of see how I could do it myself, with `MP_OBJ_IS_TYPE` and some delayed checks for number of arguments, but if it's been done before that'd be handy to reference.

dhalbert, 2020-08-11:
Maybe `vectorio/VectorShape.c`. I don't have an example of arg number checking.

tannewt, 2020-08-11:
The other reason I was ok with taking in a PWMOut was that it takes in a lot of the same parameters that we'd need for PulseOut. So, no need to duplicate the PWM parameters.

tannewt, 2021-05-26:
This came up here from @flameeyes https://flameeyes.blog/2021/05/25/whats-up-with-circuitpython-in-2021/

I think we should either switch to the (pin, ...) API fully or switch ESP32-S2 to the pwmout version. Having a mix of both is the root problem. The latter is much simpler and would cost a timer but I think is fine. I think we should switch the S2 to PWMOut for 7.x.

hierophect, 2021-05-26:
I still vote for switching entirely to a pin API, since the timers on the ESP32-S2 are so limited, and I don't like the idea of our API implying a very different internal process than what's actually happening.

tannewt, 2021-05-26:
Our API abstracts away the internals deliberately so that code works across ports. This is an example where we didn't do that and it caused non-portable code.

hierophect, 2021-05-26:
That's true, and I agree this is a tricky situation. But just playing devils advocate here, I don't think we have existing precedent for a module tying down resources from an unrelated chip feature just to fit the API, either. If that resource was in abundance, I wouldn't worry about it, but there are only 4 timers on an ESP32-S2, and I don't think it would be that out of left field for someone to wonder why their 4 servo robot is conflicting with their IR controller when Arduino and the IDF have no such conflict.

dhalbert, 2021-05-26:
I think the pin API for `PulseOut` is the best choice in the long run. We can deprecate the non-pin API in 7.0.0 and drop it in 8.0.0, or be more abrupt.

Flameeyes, 2021-05-26:
From an user perspective, there's also one important note: `PulseOut` does not declare itself as being _solely_ an IR tool.

There's plenty of other places where you may want to use pulse trains without using IR, for instance if you're trying to bitbang a serial protocol (I have in front of me a device that needs a 104bps serial… which might be easier in some cases to bitbang than hope that the hardware supports such low baudrate).

In my case, I am actually using this for a remote control protocol, but skipping the whole IR part, and just connecting to the IR decoder signal lines, so what I've been trying to do is to get the pulse out _without_ the carrier signal.

For those uses, the pin-based interface would definitely feel easier to use, as long as there's a way to disable the carrier signal. Otherwise we're back to square one.

tannewt, 2021-05-26:
> when Arduino and the IDF have no such conflict.

We shouldn't design our APIs for people who have used Arduino, the IDF or read the datasheet. Our APIs should be freestanding and *not* require underlying knowledge of chip features.

Looks like I'm outvoted on next steps so let's plan to deprecate the PWMOut version in 7.x.

> I think the pin API for `PulseOut` is the best choice in the long run. We can deprecate the non-pin API in 7.0.0 and drop it in 8.0.0, or be more abrupt.

Ok, then we'd need to:
- [ ] Update all of the other ports for the new API.
- [ ] Update all libraries (probably only IRRemote) for the new API.
- [ ] Prep to update all example code.

dhalbert, 2021-05-26:
> For those uses, the pin-based interface would definitely feel easier to use, as long as there's a way to disable the carrier signal. Otherwise we're back to square one.

Could you take a look at the ESP32-S2 RMT peripheral and see if it meets your various use cases, including no carrier and low baud rate?

Flameeyes, 2021-05-26:
> Could you take a look at the ESP32-S2 RMT peripheral and see if it meets your various use cases, including no carrier and low baud rate?

Well, I hadn't heard of the RMT peripheral before today, but as it turns out I do have a long weekend coming up… so yeah I guess I can give it a bit of a deep dive and see what I find.

(On classic ESP32 I _think_ the serial can't do the low-bps — comments I found suggest to use the soft-uart for that, but I have not gone far enough down that road to know for sure.)

Flameeyes, 2021-05-26:
> * [ ]  Update all libraries (probably only IRRemote) for the new API.

If you can provide a way somewhere to know that the new API needs to be used, I can also add it to pysdrc's Circuit Python wrapper (https://github.com/Flameeyes/pysdrc/blob/main/pysdrc/circuitpython/transmitter.py). Otherwise I guess I should just update that to have a cut-off for the older CP instead.

tannewt, 2021-05-26:
> > Looks like I'm outvoted on next steps so let's plan to deprecate the PWMOut version in 7.x.
> 
> We can discuss this in a meeting too. We could also allow both possibilities in the constructor to make it backwards compatible.

Ya, I think we should in 7.x and then remove the PWMOut in 8.x.

> If you can provide a way somewhere to know that the new API needs to be used, I can also add it to pysdrc's Circuit Python wrapper [...]

Generally we've been doing one major release with both options and then removing the old option in the next major release. This is how we did the `pulseio.PWMOut` -> `pwmio.PWMOut` move for example. So, for code that knows what is available, you'd ideally use the major version number. This won't work on ESP32-S2 in this case though because we didn't do this policy.

Flameeyes, 2021-05-28:
> RMT

Need to try the actual code, but looking at the [documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/peripherals/rmt.html) yes this should be perfectly fine to use: the carrier is _optional_ and controlled by `carrier_en`. Which is exactly what I need in my cases: RMT without a carrier.

(Having a way to have PulseOut without a carrier is literally what I need, and it sounds like, if this is exposed correctly, it would be an even better fit for my code than the M4.)

dmcomm, 2021-07-04:
PulseOut without a carrier would also be useful on the PWM-based boards. (Creating a 100% duty cycle PWMOut is wasteful and limits the timing precision.)

dmcomm, 2021-07-19:
Using the old API, some output is produced in the time between creating the PWM object and creating the PulseOut object. Will the new system prevent that?

tannewt, 2022-08-09:
What if we just construct the PulseOut as needed rather than requiring `.frequency`?

PR #6741 (2022-08-11): Remove PWMOut parameter to PulseOut
Fixes #3264
commit 35f3773e (2022-08-11): Remove PWMOut parameter to PulseOut

Fixes #3264
closed 2022-08-11 by dhalbert