adafruit/circuitpython #3265: ulab missing cholesky factorization

kevin-tracy, 2020-08-11:
The Cholesky factorization is present in micropython ulab:

https://github.com/v923z/micropython-ulab/blob/master/code/linalg/linalg.c#L93

but not present in circuitpython ulab. This is an issue because the circuitpython ulab docs include the Cholesky factorization:

https://circuitpython.readthedocs.io/en/latest/shared-bindings/ulab/linalg/index.html

dhalbert, 2020-08-17:
@jepler @v923z Did you see this?

v923z, 2020-08-17:
@kevin-tracy https://github.com/v923z/micropython-ulab/pull/170 fixes the issue. This was a problem even in `micropytho-ulab`, but since no one used the function, no one complained. Sorry for the trouble.

@dhalbert @jepler Can you update your pointer on the adafruit side? I can generate a release tag, if you want that.

jepler, 2020-08-17:
Hi @kevin-tracy and thanks for your bug report.  Can you let us know a little more about your setup?  What version of CircuitPython are you using, what board is it on, and what steps did you follow to produce the problem?

I tried 5.3.1 on a Grand Central M4, and the function is present.  However, I did not test its functionality.
```
Adafruit CircuitPython 5.3.1 on 2020-08-17; Adafruit Grand Central M4 Express with samd51p20
>>> from ulab.linalg import cholesky
>>> cholesky
<function>
```

kevin-tracy, 2020-08-18:
Hi @jepler, thank you for looking into this. I actually just got a Grand Central M4 and with CP 5.3.1 it worked for me as well. Thank you.

jepler, 2020-08-19:
I'm glad it's working for you now.  I'll go ahead and close this issue.  Please feel free to contact us on the forums, discord, or open a new issue in the future.

closed 2020-08-19 by jepler