adafruit/circuitpython #3266: board.SPI().try_lock() is not freed over soft-reset

jepler, 2020-08-11:
```
Adafruit CircuitPython 6.0.0-alpha.2-190-g817a52228 on 2020-08-11; Particle Xenon with nRF52840
>>> import board
>>> board.SPI().try_lock()
True
>>> 
soft reboot
...
Adafruit CircuitPython 6.0.0-alpha.2-190-g817a52228 on 2020-08-11; Particle Xenon with nRF52840
>>> import board
>>> board.SPI().try_lock()
False
```

jerryneedell, 2020-09-16:
I tried this on a feather_m4_ express and reproduced the problem with  CircuitPython 6.0.0-alpha.3-106-g66cf6c46c
BUT
I updated the libraries to the most recent (9/15 bundle) and the problem went away.
This is puzzling since adafruit_busdevice has not bee updated recently.....
adafruit_rfm69 has, but it is not clear how that could impact this issue...

I'll update to current main and try it a few more times ,,,,

jerryneedell, 2020-09-16:
Also tried on teensy4.1 -- same result
```

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 6.0.0-alpha.3-165-g9256e6b37 on 2020-09-11; Teensy 4.1 with IMXRT1062DVJ6A
>>> import board
>>> board.SPI().try_lock()
True
>>> 
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Hello World!

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 6.0.0-alpha.3-165-g9256e6b37 on 2020-09-11; Teensy 4.1 with IMXRT1062DVJ6A
>>> import board
>>> board.SPI().try_lock()
False
>>> 
>>> 
```

with current main
```
ress any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 6.0.0-alpha.3-267-gd774678a0 on 2020-09-16; Teensy 4.1 with IMXRT1062DVJ6A
>>> import board
>>> board.SPI().try_lock()
True
>>> 
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Hello World!

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 6.0.0-alpha.3-267-gd774678a0 on 2020-09-16; Teensy 4.1 with IMXRT1062DVJ6A
>>> import board
>>> board.SPI().try_lock()
True
>>> 

```

Shall I go ahead and close this?

tannewt, 2020-09-16:
Yup! #3378 fixed this.

closed 2020-09-16 by tannewt