adafruit/circuitpython #3270: myp-cross make TypeError

jisforjt, 2020-08-11:
Every time I try to make mpy-cross I get a type-error. It seems that micropython was getting a similar error a few years back. They have an issue on their [GitHub repository](https://github.com/micropython/micropython/issues/3196).
```
(base) user@mycomputer mpy-cross % make
Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
Traceback (most recent call last):
  File "../py/makeversionhdr.py", line 107, in <module>
    make_version_header(sys.argv[1])
  File "../py/makeversionhdr.py", line 72, in make_version_header
    git_tag, git_hash, ver = info
TypeError: cannot unpack non-iterable NoneType object
make: *** [build/genhdr/mpversion.h] Error 1
```
OS: macOS Catalina v10.15.6
Program: Terminal

dhalbert, 2020-08-11:
The fix mentioned in the MicroPython issue has already been incorporated into our sources.

What version of python are you using, and did you clone or fork our repo in a conventional way?

jisforjt, 2020-08-12:
I downloaded the repository as a zip file directly from your GitHub repository.

I updated pip (v20.2.2), git (v2.24.3), homebrew (2.4.11), and python (v.3.7). 

I retried the unzipped repository again and received the same error message.

Then I cloned the repository from git and received this error when I tried to make it:

```
(base) shadow@pypict mpy-cross % make
Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
QSTR updated
Traceback (most recent call last):
  File "../py/makeqstrdata.py", line 24, in <module>
    import huffman
ModuleNotFoundError: No module named 'huffman'
make: *** [build/genhdr/qstrdefs.enum.h] Error 1
make: *** Deleting file `build/genhdr/qstrdefs.enum.h'
```
I then tried: [source](https://forums.adafruit.com/viewtopic.php?f=60&t=143530)
```
git submodule update --init --recursive
```
This time I received this:
```
Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
__TEXT	__DATA	__OBJC	others	dec	hex
192512	4096	0	4295061504	4295258112	100047000
```
After all that it doesn't work.
```
(base) shadow@pypict mpy-cross % mpy-cross -h  
zsh: command not found: mpy-cross
```
Then I realized it copied and pasted it wrong.
```
(base) shadow@pypict mpy-cross % ./mpy-cross -h
...
...
```
__Yay, it works!__

Thanks for your assistance.

dhalbert, 2020-08-12:
Great! Glad you got it working. If you have not read this already, it may explain more for you: https://learn.adafruit.com/building-circuitpython

closed 2020-08-12 by dhalbert