adafruit/circuitpython #3271: No NEOPIXEL alias for GPIO18 on espressif_saola_1_wroom ( vs. wrover )

d-c-d, 2020-08-12:
I am very happy to have NEOPIXEL working ( thank you all), but noticed this unexpected difference between the WROOM and the WROVER boards 

No NEOPIXEL in dir(board)

Adafruit CircuitPython 6.0.0-alpha.2-216-gbbac68e77 on 2020-08-11; Saola 1 w/Wroom with ESP32S2
>>>import board
>>> dir(board)
['__class__', 'IO0', 'IO1', 'IO10', 'IO11', 'IO12', 'IO13', 'IO14', 'IO15', 'IO16', 'IO17', '**IO18**', 'IO19', 'IO2', 'IO20', 'IO21', 'IO26', 'IO3', 'IO33', 'IO34', 'IO35', 'IO36', 'IO37', 'IO38', 'IO39', 'IO4', 'IO40', 'IO41', 'IO42', 'IO43', 'IO44', 'IO45', 'IO46', 'IO5', 'IO6', 'IO7', 'IO8', 'IO9', 'RX', 'TX']
 
in circuitpython/ports/esp32s2/boards
$ git blame espressif_saola_1_wroom/pins.c | grep GPIO18
6aaab005c5 (Scott Shawcroft 2020-04-17 16:23:28 -0700 24)     { MP_ROM_QSTR(**MP_QSTR_IO18**), MP_ROM_PTR(&pin_GPIO18) },

but the WROVER does have a NEOPIXEL alias for GPIO18

$ git blame espressif_saola_1_wrover/pins.c | g GPIO18
6aaab005c5 (Scott Shawcroft 2020-04-17 16:23:28 -0700 24)     { MP_ROM_QSTR(**MP_QSTR_IO18**), MP_ROM_PTR(&pin_GPIO18) },
1c1df053d5 (Lucian Copeland 2020-07-31 14:50:18 -0400 46)     { MP_ROM_QSTR(**MP_QSTR_NEOPIXEL**), MP_ROM_PTR(&pin_GPIO18) },

tannewt, 2020-08-12:
@d-c-d Want to add it? It should be straightforward.

hierophect, 2020-08-12:
Slipped my mind when I added the mpconfig for the wroom, my bad.

PR #3280 (2020-08-17): ESP32-S2: Add Neopixel to Wroom
Resolves #3271
closed 2020-08-17 by tannewt