adafruit/circuitpython #3272: Getting AttributeError, but all libraries are copied onto CircuitPython?

Tomblarom, 2020-08-12:
I'm using CircuitPython on ESP8266 and I know it's depreciated, but I wanted to give it a shot. I'm trying to use the [Adafruit 16x2 Character LCD + Keypad](https://www.adafruit.com/product/1109) with a [Wemos D1 mini (ESP8266)](https://www.makershop.de/plattformen/d1-mini/). I installed the latest version by following [this guide](https://learn.adafruit.com/welcome-to-circuitpython/circuitpython-for-esp8266): 
`Adafruit CircuitPython 3.1.2 on 2019-01-07; ESP module with ESP8266` - `adafruit-circuitpython-feather_huzzah-3.1.2.bin`

I copied the needed libs, [using ampy](https://learn.adafruit.com/micropython-basics-load-files-and-run-code/file-operations). This is my lib directory on the ESP8266:
```
.
└── lib/
    ├── adafruit_bus_device/
    │   ├── __init__.py
    │   ├── i2c_device.py
    │   └── spi_device.py
    ├── adafruit_character_lcd/
    │   ├── character_lcd.py
    │   ├── character_lcd_i2c.py
    │   ├── character_lcd_rgb_i2c.py
    │   └── character_lcd_spi.py
    └── adafruit_mcp230xx/
        ├── __init__.py
        ├── digital_inout.py
        ├── mcp230xx.py
        ├── mcp23008.py
        ├── mcp23016.py
        └── mcp23017.py
```

Unfortunately, I'm constantly getting this Error: `AttributeError: 'module' object has no attribute 'character_lcd_rgb_i2c'`
[image]

dhalbert, 2020-08-12:
That is an odd error. Do you have an `adafruit_character_lcd` directory or .py in CIRCUITPY? It would take precedence over what's in `lib`, due to `sys.path`.

One thing though, is that you can't use a current bundle with 3.x. Many things have changed incompatibly since 3.x. You can try a much older bundle, like https://github.com/adafruit/Adafruit_CircuitPython_Bundle/releases/tag/20190111

Tomblarom, 2020-08-12:
Yes it's exactly like I illustrated above. `adafruit_character_lcd` is a folder, that's holding `.py` files.

deshipu, 2020-08-12:
Did 3.x already use the ``lib`` directory? Maybe try putting the libraries directly next to your code.

dhalbert, 2020-08-12:
`/lib` is ok. This is sys.path:
```
>>> import sys
>>> sys.path
['', '/', '.frozen', '/lib']
```
I tried copying all the .py files over to / (not /lib, that was too much trouble), and when I do the import, I run out of memory. You'll need 3.x .mpy files from an old bundle; the .py files are too big.
My guess is that there's some kind of file corruption or incomplete copy, which is causing the error you see. Try erasing everything and copying some older .mpy's instead. I was unable to duplicate that error. I did import the .mpy files and did not get the error above.

Tomblarom, 2020-08-12:
Thanks for helping @dhalbert! That worked. Used `adafruit-circuitpython-bundle-3.x-mpy-20190111.zip` from the release, you referred to. Now I'm getting another error: `ValueError: Pin GPIO4 in use`

Tomblarom, 2020-08-13:
The latest 3.x release I could find is this: https://github.com/adafruit/Adafruit_CircuitPython_Bundle/releases/tag/20190909
It worked, but I'm still getting this weird error, saying GPIO4 is in use. When the CIRCUITPY is plugged in and I reset using the button, it's outputting the following:
```
boot.py output:
main.py output:
Traceback (most recent call last):
  File "main.py", line 6, in <module>
  File "adafruit_character_lcd/character_lcd_rgb_i2c.py", line 87, in __init__
  File "adafruit_mcp230xx/mcp23017.py", line 64, in __init__
  File "adafruit_mcp230xx/mcp230xx.py", line 48, in __init__
  File "adafruit_bus_device/i2c_device.py", line 78, in __init__
  File "adafruit_bus_device/i2c_device.py", line 76, in __init__
ValueError: No I2C device at address: 20
```

dhalbert, 2020-12-18:
Closing this, since it's on an old version and is quite old now.

You might try erasing flash completely and then reloading things.

closed 2020-12-18 by dhalbert