adafruit/circuitpython #3274: PC14/PC15 unusable on STM32F411CE

sersorrel, 2020-08-12:
It looks like #2143 (specifically, 863655044e0b08a917ebaebcd6c33094391d5f87) made it impossible to use C14 and C15 on stm32f411ce-blackpill – trying to construct a DigitalInOut on either of those pins complains that they're already `in use`. [The datasheet](https://www.st.com/resource/en/datasheet/stm32f411ce.pdf) certainly suggests that you should be a bit careful using them:

>PC13, PC14 and PC15 are supplied through the power switch. Since the switch only sinks a limited amount of current (3 mA), the use of GPIOs PC13 to PC15 in output mode is limited:
>- The speed should not exceed 2 MHz with a maximum load of 30 pF.
>- These I/Os must not be used as a current source (e.g. to drive an LED).

but there are [a few](https://electronics.stackexchange.com/questions/378772/pc14-and-pc15-on-stm32f401-not-working-as-inputs) questions [elsewhere](https://stackoverflow.com/questions/57379040/unable-to-use-pc15-as-gpio-input-on-stm32f030rc) about using those pins as GPIO on similar STM32 boards, and fundamentally they appear to be meant to be usable as GPIOs on STM32F411. From looking at the PR that protected them I don't understand why CircuitPython doesn't let you.

hierophect, 2020-08-13:
PC14 and PC15 are reserved as low speed oscillator (LSE) pins on virtually all STM32 development boards. Since circuitpython depends on this oscillator for low power operation and use of the RTC, we do not provide any way to overwrite it - these pins are rarely even exposed on development boards for this reason (the Blackpill being sort of an outlier in this case).

If you urgently needed access to these pins for some reason, it is possible that we could add a flag to disable the external LSE and switch to the internal LSE, freeing them for GPIO use, but it's hard to see what the advantages of that would be if any other GPIO is available - the pins have no other useful altfunctions and have the aforementioned power concerns. I would suggest you use a GPIO expander if you require more IO, or switch to a larger package size, if you require more IO that urgently. 

All of this could be better documented, though! Lots of other vendors don't even make oscillator pins available as GPIOs so it's easy to see how it could be confusing. I'll think about where we could add extra warnings for this case.

sersorrel, 2020-08-13:
That makes sense, thanks! Yeah, being new to STM32 and to CircuitPython I basically have no idea what I'm doing :upside_down_face:

It's not a huge deal; there's at least one spare IO I can attach a bodge wire to and use instead. There are a couple of other issues with the Blackpill for this usecase too, so I think ultimately the plan is to switch to a SAMD51 board which should have regular GPIO pins in the place of the Blackpill's C14/C15.

hierophect, 2020-08-13:
Sounds good - if you'd like more help with your specific application, feel free to ask on the Adafruit Discord, there are be lots of people who would be happy to help select a chip with the right IO for whatever it is you're trying to do. You could also look into other STM32 boards that have more IO, like a Discovery board or something similar. F411 MCUs go up to 100 pin packages, with whole new banks for PDxx and PExx

Closing this issue since it sounds like a workaround is acceptable.

closed 2020-08-13 by hierophect