adafruit/circuitpython #3278: Unify ORDEREDDICT setting.

tannewt, 2020-08-14:
`MICROPY_PY_COLLECTIONS_ORDEREDDICT` is set individually for every port. We should centralize it's use to `py/circuitpy*` settings. `MICROPY_CPYTHON_COMPAT` is an example of a setting that has already been centralized.

```
../../ports/unix/mpconfigport.h
102:#define MICROPY_PY_COLLECTIONS_ORDEREDDICT (1)

../../ports/nrf/mpconfigport.h
63:#define MICROPY_PY_COLLECTIONS_ORDEREDDICT       (1)

../../ports/mimxrt10xx/mpconfigport.h
43:#define MICROPY_PY_COLLECTIONS_ORDEREDDICT          (0)

../../ports/stm/mpconfigport.h
33:#define MICROPY_PY_COLLECTIONS_ORDEREDDICT       (1)

../../ports/atmel-samd/mpconfigport.h
46:#define MICROPY_PY_COLLECTIONS_ORDEREDDICT          (0)
81:#define MICROPY_PY_COLLECTIONS_ORDEREDDICT          (1)

../../ports/litex/mpconfigport.h
33:#define MICROPY_PY_COLLECTIONS_ORDEREDDICT  (1)

../../py/modcollections.c
37:    #if MICROPY_PY_COLLECTIONS_ORDEREDDICT

../../py/map.c
175:                #if MICROPY_PY_COLLECTIONS_ORDEREDDICT
190:        #if MICROPY_PY_COLLECTIONS_ORDEREDDICT

../../py/mpconfig.h
1002:#ifndef MICROPY_PY_COLLECTIONS_ORDEREDDICT
1003:#define MICROPY_PY_COLLECTIONS_ORDEREDDICT (0)
```

gamblor21, 2020-10-13:
I can take a run at fixing this, but looking over it I am a bit concerned about the mimxrt10xx port fix. Currently the other ports define and use `CIRCUITPY_FULL_BUILD = 0` (or 1) and that is used to set other values such as `MICROPY_CPYTHON_COMPAT`.

The mimxrt10xx does not set full build, so it defaults elsewhere to being defined to 1. If we set `ORDEREDDICT = CIRCUITPY_FULL_BUILD ` it will include this feature for that port when it should not. If we define `CIRCUITPY_FULL_BUILD` to 0 then many previously included features will be excluded.

It can probably be defined separately for mimxrt10xx but then we are getting away from centralizing the define.

tannewt, 2020-10-13:
@gamblor21 I think it's fine to turn on ORDERDDICT for iMX. It's alpha and can be changed.

utkarsh147-del, 2020-11-02:
Please assign this issue to me.I want to work on this.

jepler, 2020-11-02:
@utkarsh147-del Thank you for offering!  I have assigned this to you, but I think that it may already have been resolved by #3554.  If so, I hope you'll consider working on another issue.

tannewt, 2020-11-02:
Yup, this has been done already. Sorry for the issue being open!

closed 2020-11-02 by tannewt