adafruit/circuitpython #3293: Sparkfun Thing plus SAMD51 SDA/SCL pins inverted for board.I2C Vs busio.I2C

nezra, 2020-08-17:
Excellent work so far; one issue, using circuitpython 5 on the Sparkfun SAMD51 Thing Plus

appears to be an issue with this build; makes the examples posted online and in the docs confusing for I2C.

Using "board.I2C" the SDA labeled and schematic defined pin is being used as SCL. and SCL is being use as SDA.

Using "busio.I2C" works as expected. when you define SDA and SCL as board.SDA and board.SCL and the applicable bus lines are run as SDA and SCL.(i.e. busio.I2C(scl=board.SCL, sda=board.sda)

As the examples for I2C switch back and forth between requiring the use of Board.I2C and busio.I2C, this breaks the examples.

There's no errors; just cant find devices or communicate with I2C devices unless either the wires get flipped for board.I2C examples.

nezra, 2020-08-17:
side note on this. Both the Adafruit Motorkit and ServoKit Library uses board.I2C instead of busio.I2C if i2c is None. As such, this breaks the motorkit/servokit library examples unless you specifically define the i2c and pass it accordingly.

ladyada, 2020-08-17:
thats odd - im not seeing the same? SCL == PA23 and SDA == PA22

https://github.com/adafruit/circuitpython/blob/main/ports/atmel-samd/boards/sparkfun_samd51_thing_plus/mpconfigboard.h#L21
```
#define DEFAULT_I2C_BUS_SCL (&pin_PA23)
#define DEFAULT_I2C_BUS_SDA (&pin_PA22)
```

vs
https://github.com/adafruit/circuitpython/blob/main/ports/atmel-samd/boards/sparkfun_samd51_thing_plus/pins.c#L21
```
    { MP_ROM_QSTR(MP_QSTR_SDA), MP_ROM_PTR(&pin_PA22) },
    { MP_ROM_QSTR(MP_QSTR_SCL), MP_ROM_PTR(&pin_PA23) },
```

tannewt, 2020-08-17:
Looks like it was fixed already: https://github.com/adafruit/circuitpython/pull/3060

Please use the latest 6.0.0 pre-release. We're working on getting 6.x stable released so this won't be hit by others.

closed 2020-08-17 by tannewt

nezra, 2020-08-17:
Perfect.