adafruit/circuitpython #3294: pystack exhausted 6.0.0-alpha.2 on PyPortal

anecdata, 2020-08-17:
This is a placeholder issue. I don't have a minimal reproducible example, but alpha-2 breaks my main IoT application with `pystack exhausted`. 6.0.0-1 and prior are fine. The code makes it through all of the setup, including `displayio` and `esp32spi`, but it seems the HTTP GET triggers the error. The library refactoring underway may well help.

anecdata, 2020-08-17:
Full trace: 
```
Traceback (most recent call last):
  File "code.py", line 1336, in <module>
  File "code.py", line 532, in http_get
  File "adafruit_esp32spi/adafruit_esp32spi_wifimanager.py", line 202, in get
  File "adafruit_requests.py", line 291, in get
  File "adafruit_requests.py", line 182, in request
  File "adafruit_esp32spi/adafruit_esp32spi_socket.py", line 42, in getaddrinfo
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 604, in get_host_by_name
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 324, in _send_command_get_response
RuntimeError: pystack exhausted
```

tannewt, 2020-08-17:
I believe the latest version off main will work ok. Mind trying that? We just need to release it.

anecdata, 2020-08-17:
pystack appears to have gotten enough rest by `CircuitPython 6.0.0-alpha.2-296-g6e5c2b3fe on 2020-08-17` :-) Thanks! (Github search for "pystack" yields no code or commits)

closed 2020-08-17 by anecdata

tannewt, 2020-08-17:
Glad it worked!

I turned on pystack here: https://github.com/adafruit/circuitpython/pull/3160 and the size was bumped here: https://github.com/adafruit/circuitpython/pull/3213