adafruit/circuitpython #3296: nrf: nvm sd_flash_page_erase_sync() hangs

ricardoquesada, 2020-08-18:
tested on:
 - CLUE
 - master branch. hash: 94ecf33a5ccd3682075e7a849621148ac72142a4

steps to repro:

Open REPL and then:
```python
import _bleio
_bleio.adapter.erase_bonding()
```

It fails while trying to erase page address `0xea000`  (page number `0xea`).

circuitpython-6.0-alpha2 works Ok.
I compiled circuitpython following the instructructions from here: https://learn.adafruit.com/building-circuitpython/build-circuitpython using `BOARD=clue_nrf52840_express`

ricardoquesada, 2020-08-18:
in case it helps, I'm using gcc version: `arm-none-eabi-gcc (GNU Arm Embedded Toolchain 9-2020-q2-update) 9.3.1 20200408 (release)`

ricardoquesada, 2020-08-18:
I did a bisect and the culprit commit is this one: d83a4ac72db90253089fb33e6f01d6da8f99f8b3

I tried with both with:

* arm gcc 2020-q2-update (latest one)
* arm gcc 2019-q4-major (recommended for circuit python 5 according the adafruit documnetation)

and with both I got the error. So, it seems to be a compiler error. But until it is fixed I'd remove the `OPTIMIZATION_LEVEL = 2`

dhalbert, 2020-08-18:
Typically these errors are us forgetting to make something `volatile` or something is uninitialized. The error is hidden with less optimization. I'm not sure we've encountered an actual compiler bug. Thanks for the bisect. I'll try tracking down the issue.

ricardoquesada, 2020-08-18:
makes sense, thanks.

I debugged it a bit further, and:

  *  there is a `while (sd_flash_operation_status != SD_FLASH_OPERATION_IN_PROGRESS) {}`... and code cannot exit this loop.
  * `sd_flash_operation_status` gets updated on `SD_EVT_IRQHandler`
  * but `SD_EVT_IRQHandler` is not being called after `sd_flash_page_erase()` is called. Not sure whether there is a race condition, or interrupts are masked

Might not be related to this bug, but seems to be safer to protect `sd_flash_operation_status` with an `sd_mutex` (?)

dhalbert, 2020-08-19:
`sd_flash_operation_status` is `volatile`, so that's not the issue.

We had tended not to use `sd_mutex`s because there were a number of things that happened when the SD was not enabled, but this is not one of those.

DavePutz, 2020-08-19:
Reproduced this on a Clue and other BLE functionality hangs as well. With the latest source BLE works as expected on a Circuitplayground Bluefruit. I commented out the spi and display initializations in ports/nrf/boards//clue_nrf52840_express/boards.c and then the Clue BLE works properly.
@dhalbert - do we possibly still have some memory conflict between BLE and displayio?

ricardoquesada, 2020-09-24:
should we close this since pr #3375 got merged?

tannewt, 2020-09-24:
Yup! Thanks @ricardoquesada

closed 2020-09-24 by tannewt