adafruit/circuitpython #3301: add "exceptionstring" as a member to sys

ThomasAtBBTF, 2020-08-19:
It is great that sys has:
sys.print_exception(exc, file=sys.stdout)
but it would be helpful if the exceptionstring would be available as well to process in it.

tannewt, 2020-08-19:
Is there a CPython API you are referring to? I don't see `sys.exceptionstring` here: https://docs.python.org/3/library/sys.html

ThomasAtBBTF, 2020-08-19:
Hmm,
in CPython I do:
    exc_type, exc_value, exc_traceback = sys.exc_info()
    errorlines = traceback.format_exception(exc_type, exc_value, exc_traceback)
and this will give me what I want!
Somewhere inside CircuitPython there must be code which generates "errorlines".
And these "errorlines" are printed inside "sys.print_exception"
I am looking for a way to get to these "errorlines" without the need to print them!

trailhead, 2021-04-26:
This would be handy in cases where you have a hard UART for debugging, but the REPL isn't working for whatever reason.  I'm printing my exceptions to the UART, but don't know how to get the traceback, line, or filename.

microDev1, 2021-08-08:
We now have `CPython` compatible `traceback` module implemented. Adding `format_exception()` is the next step.

closed 2021-08-09 by microDev1