adafruit/circuitpython #3329: ESP32-S2 WROVER: SD card example

emard, 2020-08-26:
Is there some recommended pinout (schematics) to
connect and example commands how to mount SD card
to Saola-1 WROVER module?

I'd like simple example how to mount SD card in 1-bit (SPI) mode
(pin saving) and 4-bit (MMC) mode (faster).

tannewt, 2020-08-26:
Hi! We haven't tested SDCards with the S2 yet. The SPI based [SD library](https://github.com/adafruit/Adafruit_CircuitPython_SD) and [it's examples](https://learn.adafruit.com/micropython-hardware-sd-cards/tdicola-circuitpython) should just work. Most pins will work up to 40mhz. You'll need to pick specific pins to get up to 80mhz. They allow for the SPI peripheral to talk to pins via the IOMUX directly.

We don't have any examples using SDIO to talk to the card and I'm not sure the ESP32-S2 has a peripheral to do it.

emard, 2020-08-27:
I have tried this pinout but it doesn't work for my ESP32-S2 and
adafruit-circuitpython-espressif_saola_1_wrover-en_US-20200827-fe73cfb.bin

gpio_csn  = board.IO34 # SD_D3
gpio_mosi = board.IO35 # SD_CMD
gpio_sck  = board.IO36 # SD_CLK
gpio_miso = board.IO37 # SD_D0

full source for my SD mount attempt is here:
https://github.com/emard/esp32ecp5/blob/master/circuitpython/sdmount.py#L1

koendv, 2020-09-12:
This is pinout and code of a board with esp32-s2, sdcard and optional lcd. HTH.

[pinout with sdcard](http://www.lilygo.cn/prod_view.aspx?TypeId=50033&Id=1321)
[pinout with sdcard and lcd](http://www.lilygo.cn/claprod_view.aspx?TypeId=21&Id=1322)
[github with code](https://github.com/Xinyuan-LilyGO/LilyGo-esp32s2-base)

jepler, 2020-09-12:
I think it's likely that it doesn't work due to the ignored value of write_value in common_hal_busio_spi_read, an esp32 flavored version of #3176.   This function (at least handling the all-ones value 0xff) is required for SPI SD cards to work.

jepler, 2020-09-13:
Closing again in favor of #3396

closed 2020-09-13 by jepler