adafruit/circuitpython #3334: ESP32-S2: Null reference exception somewhere in DisplayIO

hierophect, 2020-08-26:
As noted during testing by @ladyada, the ESP32 encounters an error when attempting to connect to an SSD1306 I2C screen using the following DisplayIO sketch (tested on Saola 1 Wrover devboard):

```
import board
import busio
import displayio
import adafruit_displayio_ssd1306
displayio.release_displays()
print(dir(board))
i2c = busio.I2C(board.IO8, board.IO9)
display_bus = displayio.I2CDisplay(i2c, device_address=0x3d)
display = adafruit_displayio_ssd1306.SSD1306(display_bus, width=128, height=64)
```

The debug console for the Saola indicates that this is likely a null dereference exception, as per `EXCVADDR` being 0: 
```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4001abc4  PS      : 0x00060c30  A0      : 0x800b20fe  A1      : 0x3ffde130  
A2      : 0x3ffda068  A3      : 0x00000000  A4      : 0x00000068  A5      : 0x3ffda068  
A6      : 0x00000020  A7      : 0x00000006  A8      : 0x00000060  A9      : 0x3ffde060  
A10     : 0x3ffc6a20  A11     : 0x3ffde0ec  A12     : 0x3ffde0e8  A13     : 0x3ffde040  
A14     : 0xfffffff4  A15     : 0x3ffde040  SAR     : 0x0000001e  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x3ffde0e8  LEND    : 0x3ffde040  LCOUNT  : 0x40028fc1  
```

While discovered in #3315, this issue is also present in main and so should now be tracked here.

hierophect, 2020-08-27:
Through repeated use of ESP32 log messaging I've pinned down the null dereference to this exact point: https://github.com/adafruit/circuitpython/blob/24fb08dbd9a16e83aded3f0874bc96fcd0462fea/py/runtime.c#L1527, but... there's nothing there. The pointers that go into it are valid, and the call itself is actually just an inline macro with nothing interesting about it. https://github.com/adafruit/circuitpython/blob/24fb08dbd9a16e83aded3f0874bc96fcd0462fea/py/nlr.h#L144

I thought this might be the fault of a a deterministically placed interrupt routine that just happens to land here, but moving the actual log message around causes a failure wherever you put it, so I'm confident that isn't the case. I'm still stumped on why exactly it fails within the nlr_raise when all the parameters are valid.

DavePutz, 2020-09-07:
For what it's worth, I have been unable to reproduce this issue on a Saola 1 Wroom devboard.

tannewt, 2020-09-08:
@hierophect Where are you at on this? Has it inadvertently been fixed?

Thanks for the retest @DavePutz

hierophect, 2020-09-09:
@tannewt I'm getting back to it today, I took a break from banging my head against it at the end of last week to do some F1 stuff and cool off. Even if it's been inadvertently been fixed I'd like to figure out what it was - it seemed related to memory corruption so even if it disappears here I'd worry about it popping up somewhere else.

closed 2020-09-14 by tannewt