adafruit/circuitpython #3339: ESP32-S2: Is SPI still working?

emard, 2020-08-27:
I have ported a small project for ECP5 JTAG programmer
from micropython to circuitpython:
https://github.com/emard/esp32ecp5/tree/master/circuitpython

Last known good circuitpython that runs it correctly was this:

adafruit-circuitpython-espressif_saola_1_wrover-en_US-20200827-fe73cfb.bin

In later versions 2MB PSRAM support appears which is absolutely great,
but now SPI is either not working or has issues if initialized and deinitialized often.

This project needs to switch between bitbanging of JTAG pins and 
driving the same pins by hardware SPI for speed. JTAG traffic
doesn't always consist of multiple of 8 clocks so SPI can't be constantly
used.
During the switch from bitbanging to SPI and back,
glitches appear (pins change state while
we didn't want them to).

Some glitches are tolerable and some must be workarounded so
the project is sensitive to what is done with the pins.

tannewt, 2020-08-27:
What happens with SPI? It's possible the external RAM prevents DMA.

(I have a JTAG library started here: https://github.com/adafruit/Adafruit_CircuitPython_JTAG based on tinyfpga's programmer for the Mach XO2)

emard, 2020-08-27:
I'd say SPI doesn't work at all now :) but I don't know the details,
for this I must analyze and investigate more.
Apart from glitches, what I see currently that it SPI doesn't send correct
data I pass it (probably it sends all 00 or FF) and reads all 0. Only the
bitbanging part is still working.

This project esp32ecp5 is in use whole year for micropython ESP32-WROOM for
ECP5 and now I got it also to work on ESP32-S2, almost at the same speed as
on micropython ESP32 dual core.

tannewt, 2020-08-28:
Ok, we'll need to retest SPI. Please let me know if you have a simple test case that fails.

emard, 2020-08-28:
SPI SD card won't mount.
I tried SD card with:
adafruit-circuitpython-espressif_saola_1_wrover-en_US-20200827-fe73cfb.bin
adafruit-circuitpython-espressif_saola_1_wrover-en_US-6.0.0-alpha.3.bin
the code that is trying to mount the card:
https://github.com/emard/esp32ecp5/blob/master/circuitpython/sdmount.py#L1
output:
import sdmount
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "sdmount.py", line 30, in <module>
  File "adafruit_sdcard.py", line 116, in __init__
  File "adafruit_sdcard.py", line 177, in _init_card
  File "adafruit_sdcard.py", line 145, in _init_card
OSError: no SD card

emard, 2020-08-29:
Another even simpler test: SPI LOOPBACK, connect MISO-MOSI IO35-IO37

adafruit-circuitpython-espressif_saola_1_wrover-en_US-20200827-fe73cfb.bin passed this test

adafruit-circuitpython-espressif_saola_1_wrover-en_US-20200829-d858d04.bin
fails this test

https://github.com/emard/esp32ecp5/blob/master/circuitpython/spiloopback.py#L1

tannewt, 2020-08-31:
@emard Can you confirm it works for you with psram disabled? I think we're trying to dma from memory that can't be used for DMA.

emard, 2020-08-31:
where is download for non-PSRAM circuitpython?

I have only WROVER-S2 with PSRAM and
circuitpython binary automatically enables it.

tannewt, 2020-08-31:
@emard try loading the WROOM version. It should work just fine and not use the PSRAM.

emard, 2020-08-31:
spi loopback works with

adafruit-circuitpython-espressif_saola_1_wroom-en_US-20200831-877a4f4.bin

which doesn't work with PSRAM version

but I can't test real thing ecp5.py because
out of memory at wroom circuitpython

tannewt, 2020-09-01:
Ok, thanks for testing this! I've gotta decide how to fix it. We want bytearrays in PSRAM by default but need them in RAM when sending them to SPI.

emard, 2020-09-01:
I need min buf size of 4K. If I can have 16K there's small performance gain

python maybe needs help to place bytearray in RAM instead of PSRAM.
buf=machine.ram(bytearray(512)) or similar... but that are just my random
thoughts, any fix is good thing :)

tannewt, 2020-09-02:
Do you reuse the buffer? Would it work if we moved the bytearray into RAM when first used for DMA?

tannewt, 2020-09-02:
Reusing it is definitely the way to go.

I don't think there is a way to currently get this to work. It requires some changes under the hood to move memory as needed. Hopefully it'll just work in the future with a copy cost on first DMA.

emard, 2020-09-03:
Ok, let me know when I can pull compiled binary and test.
In the meantime I took WROVER-E with micropython and
tried my source with 256KB buffer which is required to hold
content of erase block of new FLASH chips and it works good,
and with some speedup as buffer is larger.

I don't know how they make this work, is old ESP32
better handling PSRAM DMA but I guess WROVER-E
with all loaded is not likely to have 256KB free in RAM.

emard, 2020-09-07:
I tried current binary but it still doesn't work,
seems SPI-RAM-DMA fixes are not yet there,

I have tried this:

adafruit-circuitpython-espressif_saola_1_wrover-en_US-20200906-a9779b9.bin

tannewt, 2020-09-08:
My plan is to work on it tomorrow. I'll update here when a fix is created.

tannewt, 2020-09-10:
Ok, I've confirmed its very fast at transmitting zeroes. :-) Working on a fix now.

PR #3393 (2020-09-17): Add non-DMA SPI support.
This fixes SPI with PSRAM allocated buffers. DMA with SPI2 was
attempted but produced junk output. This manual copy is less than
2x slower than DMA when not interrupted.

Fixes #3339
commit c7c90f47 (2020-09-17): Add non-DMA SPI support.

This fixes SPI with PSRAM allocated buffers. DMA with SPI2 was
attempted but produced junk output. This manual copy is less than
2x slower than DMA when not interrupted.

Fixes #3339
closed 2020-09-17 by tannewt